jpconstantineau/BlueMicro_BLE #16: low power optimization

matthewvilim, 2018-11-19:
Would it be possible to reduce power by using interrupts on the GPIO pins instead of polling? Is this something you've tried but didn't work or haven't bothered to optimize yet? Do you know if the nRF52 and BlueFruit library would support this? Sorry if this is a duplicate of previous discussion; I couldn't find anything.

jpconstantineau, 2018-11-19:
I agree with you that using interrupts and essentially have the CPU
sleeping would reduce current consumption significantly.

Because we have a matrix of keys with diodes between rows & columns, we
can't just listen to all GPIOs and pick-up specific keypresses on
interrupts.  This works well when we have 1 key per GPIO, however, not many
keyboards are implemented that way.
From what I understand, the Mitosis is.  Most keyboards implement all the
keys through a row/column matrix.  This helps maximize the number of keys
with a given number of GPIOs.

I did look into current consumption a while ago and did a number of
"experiments" while measuring current consumption any noted that peak
current consumption is when the radio runs, followed by when the CPU is
busy running code.
One element to reduce current consumption with the firmware is to carefully
choose the delays that are used in the matrix scanning routine.  This
significantly reduced current consumption.

Once we have migrated to BSP 0.9.0, and that a few important features are
implemented, I will have a look again at current consumption and how we can
reduce it further.  I have a couple uCurrent Gold and a digital
oscilloscope to measure these currents.

If you know of a method to merge interrupts with matrix scanning let me
know... I sure would look seriously into it.  Right now, all I can think of
is to use interrupts between "long" periods between user typing intervals
and standard scanning when the user types.  The interrupts would
essentially detect when the user starts to type and resume scanning.  After
a timeout once the user stops typing, the firmware would  switch the
scanning routine to "sleep" mode until the next interrupt.  This would
essentially be a hybrid method of key-scanning.

I haven't tested any of this and don't know if it's feasible.  Let me know
if you would be willing to investigate on this...

matthewvilim, 2018-11-19:
Got it. The keyboard I'm working on doesn't use a matrix layout. Your idea seems plausible; if the wake up is quick enough, you might be able to squeeze in a matrix scan immediately and figure out which keys were pressed. Do you know the ballpark numbers for wakeup time vs a matrix scan by any chance?

I'll start by trying to set up interrupts when my board arrives with a GPIO per  key and see how much power we can actually save.

jpconstantineau, 2018-11-19:
If you want to work on a branch to bring in support for per-key GPIOs, I
would welcome it.
My approach for this would be to use one of the unused GPIOs for a single
row and use the columns for all the different keys.
This way, you could test the board to make sure it works, then see how to
migrate towards using the GPIOTE or GPIO nrf52 peripherals.

One note: we are making significant changes to the key.cpp file.  It
doesn't do the scanning itself but may impact how things can be done...

Another element to think about: how can we do debouncing with interrupts?

See here for some information on the two NRF52 peripherals  (
http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Fgpio.html&cp=2_1_0_19&anchor=concept_zyt_tcb_lr
)

jpconstantineau, 2018-12-15:
Due to the way we handle double tap, long presses and other time-dependent keymap functionality, we need to call the scanmatrix routine regularly.  As such, interrupt-driven keypresses may be difficult to implement.

I did a lot of testing for power consumption and I was able to bring down consumption to just below 1mA when connected to a computer.  When not connected, there is a timeout where the whole board shuts down and current consumption essentially goes to 1uA.  It can be awaken by pressing any key and it reconnects to the computer, if the computer is paired.  Wake up takes about a second. Something I feel would be ok when a computer comes back from sleep but not for when the user actively uses the computer.

I'll close the issue for now since we have found a reasonable implementation to get good battery life.

closed 2018-12-15 by jpconstantineau