jpconstantineau/BlueMicro_BLE #33: Implement LEDs for caps, num and scroll lock

jpconstantineau, 2018-12-15:
This requires using the setkeyboardledcallback in the BLEHIDadafruit service.

This is a relatively Low priority item.

jpconstantineau, 2019-10-20:
LEDs can be set using the callback set.  
Since not all hardware has LEDs, it's not really a priority

jpconstantineau, 2019-12-05:
This has been done. 
Works as expected using the standard adafruit LEDs.

closed 2019-12-05 by jpconstantineau

laserscout, 2022-11-22:
Short question @jpconstantineau as I didn't find a clear answer elsewhere.
Besides having the indicators, does BlueMicro also **read** the caps, num, scroll lock status via **BLE**?

Thanks

jpconstantineau, 2022-11-22:
Yes, in a way...  The callback returns an integer where the bits are mapped to specific items.

See [here]((https://github.com/jpconstantineau/Community_nRF52_Arduino/blob/4dacbfabb170843c275c4afaac6f8e2c08d1b319/libraries/Bluefruit52Lib/examples/Peripheral/hid_keyboard/hid_keyboard.ino#L129)