jpconstantineau/BlueMicro_BLE #37: Add support for WS 2812b RGB leds

jpconstantineau, 2018-12-15:
Use I2S nrf52 peripheral to send RGB LED config data serially to the strip of LEDs

There are a few articles out there that have shown that I2S can be used to push RGB data to the LEDs without the need to use CPU bit banging.

The timing for the serial data needs to be quite precise.  Bit banging using the cpu would consume lots of cpu cycles (and power) and would likely fail since precise timing is needed. FreeRTOS implementation of the keyboard code and the need for the softdevice to handle BLE comms would regularly interrupt the bit banging stream and sending the LED config would fail.

Additionally, voltage needed for running the RGB LEDs has a minimum of 3.6V. this means that we would need to connect VCC of the RGB strip directly to the battery.

A MOSFET inline with VCC powering the RGB strips may be needed to eliminate the large quiescent current needed to power the LEDs, even when they don't light up.

Hardware testing not complete.

marhalloweenvt, 2019-02-23:
I have tried to add these code https://www.tweaking4all.com/hardware/arduino/arduino-all-ledstrip-effects-in-one/ into firnware_main.cpp and flash my nrf52832 development board. WS2832b effects work perfectly , but there are some errors/issues that i can't figure out. Keyboard's still recognized by my computer but there are no key has been sent from nrf52832.
Could you advise me something to workaround with this? At this time, i just use one side keyboard/master mode.

jpconstantineau, 2019-02-23:
I suspect that these two libraries use bit-banging to send the data to the
neopixel/ws2812. This essentially "locks" the CPU in doing a single task.

Some people have used the i2s peripheral of the nrf52 to take the task of
sending the data away from the CPU.

Your implementation would be useful for testing that the ws2812 hardware
works fine. Once the hardware works, one has to get the i2s peripheral
configured and working to send the data.  If you are interested in working
on this, I can send you a link to a couple of webpages that explain the
process.

marhalloweenvt, 2019-02-23:
Thank you. My hardware, circuit are very simple. I use a ws2812b 4x4 module from China to test this out.
I am very appreciated that you would send me some articles (document, blog, web pages,...) that can help me work around with this issue. I will try my best

jpconstantineau, 2019-02-23:
Here are a few links:

https://electronut.in/nrf52-i2s-ws2812/

http://takafuminaka.blogspot.com/2016/02/nrf52832-ws2812b-5-i2s.html

https://github.com/takafuminaka/nRF52832/tree/master/i2s_ws2812b_demostration_planB

The first one is in english and can provide a lot of context on how this
works.

I also added into develop the small arduino project I started a long time
ago to test this out.  Look for \BlueMicro_BLE\arduino_tests\I2S_tester

marhalloweenvt, 2019-02-23:
Ok, thanks. It will take time but i will try my best and report my result

jpconstantineau, 2019-06-07:
Hi.  Have you had a chance to look at the WS2812 with the BlueMicro?

marhalloweenvt, 2019-06-10:
> Hi. Have you had a chance to look at the WS2812 with the BlueMicro?

Hello, recently, because of work, i traveled a lots, so i haven't had time for this hobby. At last time i tried, i can set static color but not the animation. There should be some issues/conflictions. I will be back to my work after these seasons. Hope it will be soon.

Regards,

jpconstantineau, 2019-10-20:
Neopixel library works very well with the nrf52832.  I tried it and it works.  I looked int the code and it appears that there is even code specific for the nrf52 series of chips.  RGB support is therefore possible.  We now just need to code something to make it look cool!

jpconstantineau, 2020-04-24:
Code works fine with neopixel library.  closing

closed 2020-04-24 by jpconstantineau

j1n6, 2020-10-25:
Is the Neopixel used in the nice!nano firmware?

jpconstantineau, 2020-10-25:
It's used in the bluemicro firmware.

There is an updated documentation page in the develop branch that might
help you set it up for your keyboard.

The few lines extra will need to be added to the hardware_config.h file.