jpconstantineau/BlueMicro_BLE #50: Add shifted keycodes

ResoluteFury, 2018-12-30:
In QMK this is done by registering LShift, registering the base keycode, and unregistering LShift. If handled in the same way, this could be something to attack after #35.

Alternatively, it seems like the Adafruit nRF52 BLE library has a [lookup](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/bd0747473242d5d7c58ebc67ab0aa5098db56547/libraries/Bluefruit52Lib/src/services/BLEHidGeneric.cpp#L273) for shifted codes. Maybe the HID Keycodes could be adapted with the same?

jpconstantineau, 2018-12-31:
Julian has implemented shifted keys (and other modded keys) in his recent
work. It's in the Develop branch. It might need further documentation,
testing and perhaps creating  a bunch of preprocessor macros to make it
simpler to get directly to some if these shifted keys.

ResoluteFury, 2019-01-07:
Cool! I see the advanced_keycodes.h includes mod actions on that branch. I'll write some macros for stuff I use and see how it works.

closed 2019-06-07 by jpconstantineau