jpconstantineau/BlueMicro_BLE #60: Getting the build to work on Linux

themmes, 2019-05-09:
Hi Pierre,

Your switches arrived today! 

**Describe the bug**

**To Reproduce**

1. I soldered the switches and batteries in 
2. installed arduino-1.8.9 and cloned BlueMicro_BLE to my Linux
3. I copied "BlueMicro_BLE/firmware/keyboards/ErgoTravel” to "BlueMicro_BLE/firmware/keyboards/ErgoTravelLight” and kept contents the same for a first test
4. Now I need to build the firmware. I changed paths in “BlueMicro_BLE/build/build-ubuntu” to the following:
```
arduinoPath="/usr/local/arduino-1.8.9"
arduinoDataPath=$(cd /usr/local/arduino-1.8.9 & pwd)
nrf52PackagePath="/usr/local/arduino-1.8.9/packages/adafruit/hardware/nrf52"
```

**Expected behavior**

I expected the software to build, but got a redirection error.

**Desktop (please complete the following information):**
Macbook Pro 2012
VirtualBox Linux Xenial 16.4
Python3.6

**Additional context**

After building I need to flash the BlueMicros, this is still very much a mystery to me. How do I connect my BlueMicros to my linux environment (JLink, USB to TTL (3.3V) or Bluetooth)?

jpconstantineau, 2019-05-10:
Ubuntu build:  the script calls a specific folder that's for an old version of the Adafruit library.  When you installed the library, it would have been installed to a slightly different folder.  As such, you need to update the path to point to the right place.  If you make changes, keep track of them in GIT and you can make a PR and I'll bring the changes in.

For configuring the Adafruit library to compile on the Arduino IDE, follow the instructions to setup the Adafruit nrf52 Feather here: https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide?view=all#arduino-bsp-setup

Once the steps are done, try to compile one of the examples to be used with the nrf52 feather.  Follow the steps in this video to help: https://www.youtube.com/watch?v=0skX6HPSZnU

For the hardware: With the ErgoTravel, the BlueMicro V2.0x hardware supports LiPo batteries. All you need to do is plug a USB cable to the micro USB connector and to a power source (there is no data connection).  This will turn on the on-board LiPo charger.  LiPo batteries are often marketed as 3.7V batteries.  When charging LiPo batteries, they get charged up to 4.2V. Any voltage higher is unsafe. The on board LiPo charger makes sure that it doesn't exceed it.  

Programming the BlueMicros: Programming the bootloader must be done with a Jlink.
Programming the firmware can be done either either the jlink or a serial adapter (with the serial adapter, you need to press reset as you send the flash command).  

You can use either the scripts or the Arduino IDE to perform the flashing tasks above.  When in the Arduino IDE, you can turn on the verbose mode in the preferences.  This will print out the command line that gets called to flash.  These are the same command line commands that gets called by the scripts.

Once flashed, you will be able to "see" the BlueMicro using bluetooth.  You should be able to connect to it using  standard bluetooth tools on the OS you are using.

For the ErgoTravelLight: configuration is identical to the ErgoTravel.  You don't need to create a special folder for it.  You just need to create your own keyboard layout and compile/flash it.

jpconstantineau, 2019-05-10:
One more thing.  I program the assembled BlueMicro with a default firmware before shipping it.  As such, when you power it on, you should be able to see it in your bluetooth list on your computer.

themmes, 2019-05-11:
Thanks a lot for the extensive answer!

**Default firmware**
Thank you for mentioning this, this enabled me to connect the keyboards and test the hardware before continuing with the code.

**Ubuntu build**
Will do, I forked the project and create PRs when things are working. The `macos-build` seems to work right out of the box, so I might go with that.

**Charging**
That is very convenient! Why is the micro-USB port not used for data connection to pc (e.g. for flashing)?

**Programming the BlueMicros**
So if I understand correctly we connect the RXD, TXD, 3.3V and GND from the `USB to TTL` to the `BlueMicro`. How do you make this connection? Do you have extra sockets soldered on the PCB for these?

jpconstantineau, 2019-05-11:
For the USB data connection to work: It needs a serial chip on board: the
first version I built had one but didn't work for some reason.  I therefore
removed the chip from it.  It also reduces power consumption (since we
don't need to power that chip when just using it as a wireless keyboard).
Helps reduce assembly and costs too.

To program the BlueMicros using serial, you need to connect to the tx and
Rx lines on either the blumicro or (for the ErgoTravel) these same signals
brought out on the PCB itself. You also need to connect GND. No need to
connect vcc since both the serial adapter and the BlueMicro have their own
3.3v supplies.

When connecting serial lines from the BlueMicro to the serial adapter, make
sure you connect tx to Rx and Rx to tx (crossing the lines). Each line is
directional and one side transmits to the other side which receives...
Hence the crossing of the labels...

jpconstantineau, 2019-06-07:
Hey Tom.

Adafruit just released their BSP 0.11.1.  I have been working to patch both master and develop so that ce code can compile and flash to the nrf52832.  

Can you test the mac scripts to make sure that they all check out?
same with linux scripts...

Have you had a chance to resolve the issues above?

themmes, 2019-06-07:
Hi Pierre,

My keyboard is working fine now, I fixed the shorts and its working. I still have a few other issues, but those I will raise later.

I will soon test this however at the moment I do not have access to my old MacBook (with usb) as it died and I will go on holiday in a few hours.

When I am back the 16th I hope to have new parts for my MacBook and will test this too.

jpconstantineau, 2019-06-07:
Have a good Holiday.

themmes, 2019-06-30:
I do not have the parts to fix my macbook yet, but bought a USB adapter for my new macbook. I will soon take the time to look at your updates.

themmes, 2019-11-20:
Pierre, 

I’m so sorry I have never got back to you on this, your newsletter reminded me. I have not been able to make the ErgoTravel my daily driver. It was too laggy (there was especially a lag difference between left and right causing swapping letters) and I havent had time to work on the code too improve this.

For now I am working with a GK61 and that is doing fine for me. Hopefully in the future I’ll be able to dedicate some time for the ErgoTravel, but I do not have that opportunity right now.

All the best,
Tom

jpconstantineau, 2019-11-20:
Hi Tom

Someone had similar issues and by connecting a separate antenna and
relocating the small 0ohm resistor to connect it, he was able to improve
connectivity between the two halves.

I haven't personally tested it yet to compare performance but I recently
received a few antennas to give it a try.

Pierre

jpconstantineau, 2020-10-17:
We now have build scripts for macos as well as ubuntu and docker.  Closing.

closed 2020-10-17 by jpconstantineau