jpconstantineau/BlueMicro_BLE #61: Building for macOS

themmes, 2019-05-11:
**Describe the bug**

Building on macOS sadly is not going as smooth as  I hoped.

[image]

[image]

**To Reproduce**

1. follow `nrfutil` installation instructions
2. running `build-macos` to create ErgoTravel:default `.hex` and `.zip` in `/output/`
3. running `flash-serial-macos ErgoTravel:default:right /dev/cu.usbserial-1410` to flash BlueMicro

**Expected behavior**
I would expect a successful flash

**System (please complete the following information):**
macOS mojave 10.14
python 2.7 (and 3.7)
nrfutil==0.5.3
pip==9.0.3

jpconstantineau, 2019-05-12:
You need to reset the board to activate serial flashing as you see the
flashing command about to being called.

Why: the bootloader goes for a few moments into serial DFU. If after a
second it doesn't detect a DFU request, it goes and run the firmware.  The
firmware doesn't pickup the DFU connection and essentially ignores anything
the computer would try to do...

If you want to avoid any issues flashing with serial and timing the reset
to when it is about to start flashing, you can flash the bootloader using
the jlink.  When re-flashing the bootloader, it wipes the firmware and when
the bootloader starts, it remains in serial dfu (since it doesn't have a
firmware to jump to)

themmes, 2019-05-12:
Hmm I tried it many times now, I am getting two different error messages. I either get 
`Failed to upgrade target. Error is: No data received on serial port. Not able to proceed.` or `Failed to upgrade target. Error is: Attempting to use a port that is not open`

I'll try it with the JLink later..

themmes, 2019-05-12:
So I ran `flash-swd-macos.sh`, after installing the right dependencies (`nrfjprog`, `merge hex` and `jlinkram.dll`) some adaptation of the paths this ran perfectly. However, now there seems to be no program on the hardware? The print statements in `flash-swd-macos.sh` suggest otherwise.

So I tried running `flash-serial-macos.sh` again, but I am still getting the same error `Failed to upgrade target. Error is: No data received on serial port. Not able to proceed.`.

jpconstantineau, 2019-05-12:
To check the serial hardware and see it is works fine do the following
steps:
1- put a jumper between Rx and TX on the serial adapter (no other
connection) this passes everything sent to the receive ..
2-:using Arduino IDE, select serial adapter
3- select serial monitor of the IDE
4 a window should come up where you type at the top and you see what gets
received at the bottom
5 type anything and press enter/send
6 you should see what you typed below. if not, then there is either a
driver problem or you have selected the wrong serial port/adapter in the
menu.

Hope this helps troubleshooting your serial connection.

themmes, 2019-05-13:
I am learning by the minute!

I tested my serial and it seemed my serial connection works just fine, it seemed to be in one of my cables so swapped that one and then the flashing worked like a charm!

Now I seem to have some keys registering whole rows. Not surprising for my first soldering project... 🤫

wikkealphenaar, 2019-05-14:
(Tried to type this message on my keyboard, but because of shorts its not working) I think this issue can be closed now, building on macOS worked like a charm

themmes, 2019-05-24:
I do not get this to work anymore and there seems to be no program on my boards..

How can I see if the device is actually on? 

1. I connect the serial connector it lights either 1 or 2 lights depending on the power switch. 
2. I assume two lights means on.. 
3. Running the flash-swd does not work this way, 
```
ERROR: Recover failed. Please make sure that the correct device family is given
ERROR: and try again.
```
however when I switch back it does run
```
Recovering device. This operation might take 30s.
Erasing user code and UICR flash areas.
Erasing user available code and UICR flash areas.
Applying system reset.
Programming bootloader..Parsing hex file.
Reading flash area to program to guarantee it is erased.
Checking that the area to write is not protected.
Programming device.
Programming keyboard..Parsing hex file.
Reading flash area to program to guarantee it is erased.
Checking that the area to write is not protected.
Programming device.
Programming app_valid_setting..Parsing hex file.
Reading flash area to program to guarantee it is erased.
Checking that the area to write is not protected.
Programming device.
Applying system reset.
Run.
```
4. This probably means my assumption of on was incorrect
5. On to flash-serial, whatever I try I cannot get this to work (even though the Arduino IDE serial monitor is working fine)

themmes, 2019-05-29:
Sorry, I have spend another evening trying all different options I see to get them to flash. But it just is not working. Is there anything else I can test apart from the Serial monitor in the Arduino application?

jpconstantineau, 2019-05-29:
Let me see over the next few days.  I need to catch up on a number of
things as I was away on holiday for the better part of 3 weeks.
Let me check if things are up to date and see what versions I have so that
we can sync-up these.

I don't have a mac.  As such, we will need to go back and forth on testing
things out...

Just to be clear so that I know where to start from...
1 - you tested the serial adapter with the jumper between RX/TX and the
arduino IDE.  With the serial monitor you were able to see what you typed
back. (this tests the serial adapter and the drivers for it)
2 - you flashed the bootloader with the jlink (this tests that you can
actually flash the bootloader and any other firmware by methods other than
the serial adapter)

themmes, 2019-05-29:
Its fixed!

Steps I took:
1. Reinstall everything from `Arduino IDE` to `nrfjprog` and `mergehex`
2. Run `bootloader-macos.sh`
3. Run `flash-swd-macos.sh`
4. Run `flash-serial-macos.sh`

The main problem was I believe either an outdated bootloader or (more likely) my timing of the reset, apparently I should trigger the reset way before I start the flash process from the CLI.

closed 2019-05-29 by themmes

themmes, 2019-05-29:
@jpconstantineau Thanks for checking in again this quick after (and even during your holidays!). Its amazing what you build so far, only have macs to work from, but I hope I can contribute to the mac side of things once I have everything up and running!