jpconstantineau/BlueMicro_BLE #63: Match short in key row/column with pin on BlueMicro

themmes, 2019-05-14:
So I am having shorts in the hardware causing multiple keys to trigger an entire column. The following is with `ErgoTravel:default`.

Left

- g -> g + space (D22 + D24?)
- space -> g + space (D22 + D24?)
- a -> cmd + a (D8 + ?)
- cmd -> a +cmd (? + D8)
- s -> s + option (? + ?)
- option -> s + option (? + ?)
- f -> f +f4 (D18 + D20)
- f4 -> f +f4 (D18 + D20)

Right

- , -> + whole row

themmes, 2019-05-14:
I just fixed the issue on the right side, it was due to shorting through the metal casing of the BlueMicro..

themmes, 2019-05-22:
So I used the schematic [1] to get a sense of how every key is connected and how the keyboard matrix principle works. Then I used the PCB image [2] to translate this to the real life board.
1. https://github.com/jpconstantineau/ErgoTravel/blob/master/PCB/ErgoTravel_V1.03_Schematic.pdf
2. https://github.com/jpconstantineau/ErgoTravel/blob/master/PCB/ErgoTravelPCBV1.PNG

The problem is a connection between row1 and row3 (PB4 and PD7). I confirmed this with a multimeter. Do you have any clue where this short could be? Because the pins are not together on the BlueMicro (row2 is in between).

jpconstantineau, 2019-05-22:
Is the problem the same for both sides?
Let me check on things.  They shouldn't be shorted. Just to be sure, which
version of the board is written on there? (The ergotravel)

Did you do your continuity tests without the bluemicro in the socket?( Just
making sure I replicate your conditions)

themmes, 2019-05-23:
v1.01 for both sides, but I actually only have this short on the left board. Yes, I removed the BlueMicro from its sockets and measured the sockets.

jpconstantineau, 2019-05-23:
Ah ok.. if it's only one side, that reassures me that it's not a design
issue but one specific for the board in question.

It's also not on the bluemicro side but the keyboard PCB.  I assume you
have all switches installed by now. I also assume you have the
screws/standoffs installed since the screw heads need to be between the top
plate and main PCB.

If I remember well, I'll have to check the drawings when I get home from my
long weekend out, the diodes connect the switches to the rows. (It is two
rows or two columns that are shorted?)
I assume that you installed the diodes on the underside of the board. This
means that the legs were poking up and that you clipped them close to the
board.  What's possible is that you clipped the legs and left a little
"mountain" that too tall to fit in the narrow gap between the two boards.
The top plate has a copper fill all over to ensure the opacity of the
plates.  This means that if two such "volcano mountains" touches the top
plate and scratch the solder mask , then you may have a short between two
of the diode legs.  It's a long-shot but try to shine a light in the gap
and see if there is a potential short there...  What you could also do is
apply heat to the common side of the diode leg and with tweezers and pull
the leg out slightly to reduce the height of the clipped leg on the other
side.
Before doing so, check for continuity between the diode legs across
row/columns and that might help identify if that's the issue. Look for any
that shouldn't be connected together.(a resistance measurement might help.
The lowest reading might point to the combination with the shortest current
path.)

If continuity is on the switches side, then I can't think of what it could
be. Again measuring resistance between points might help in identifying the
shortest path.

V1.01. I have a few of these boards here.
One of the last solution I have in mind is to cut traces and place a wire
to reconnect things.  For this I will need to identify which trace location
would work best and which side.

Which half isn't working? That will help me see what traces are available
on the bottom side and which ones are sandwiched with the top plate.

themmes, 2019-05-23:
Hi Pierre,

- It is the left half
- Indeed, I have the standoffs and all switches installed

> This means that if two such "volcano mountains" touches the top
> plate and scratch the solder mask , then you may have a short between two
> of the diode legs.

This may very well be the problem, I already had to cut some down further to get the top-plate and pcb close enough for the switches to fit.

1. I'll try if I can see and else measure which of the diodes legs went through the solder mask.
2. If so I'll try to heat and lift from the common side

themmes, 2019-05-23:
Terrible, many of the "volcano"s seem to touch the top plate, I have tried to make it a bit better for as far as I could see and/or reach the diodes.

However, cannot yet test if it had any affect, because the boards are not respondent at all. Went through all the steps of #61 again, but no success this time. I thought maybe the batteries are empty by now? So I am trying if a recharge will help...

jpconstantineau, 2019-05-24:
Let me know...

With continuity tester of a multi meter, or a diode tester of a multi
meter, one can test rows /columns and keyprsses.

themmes, 2019-05-24:
Hi Pierre,

Any ideas on how to locate the short? It can be any of the diode legs (black side) for each row (row1 and row2). The resistance readings I am getting are all the same, or at least very similar, the way I touch the diode leg with my multimeter measure point matters most...

themmes, 2019-05-29:
#61 Is resolved and I was able to test my keyboard again, however the shorts still seem to be there, despite my efforts to melt some of the "volcanos" from the common side. I guess I'll be desoldering all the switches tomorrow night!

themmes, 2019-06-03:
Good news, I did not have to desolder all the switches. I managed to pull back the diodes even further from the common side. This finally fixed the shorts!

closed 2019-06-03 by themmes