jpconstantineau/BlueMicro_BLE #74: Support for nrf52840

jpconstantineau, 2019-10-20:
here are a few tasks that should be done before support for the nrf52840 is possible
1 - detect if the nrf52840 is connected to USB
2 - have the USB HID working
3 - Switch between USB HID and BLE HID automatically when connecting/disconnecting USB connection

The two two steps could be done in the arduino IDE using simple sketches.  Then we can wrap-it for inclusion withing the Firmware.  Use the "arduino_tests" folder to create such test/development scenarios.

jpconstantineau, 2019-10-20:
For   Point #2, we can use the  tinyusb portions included in the core [here](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/200b3aaefb3256ac26df82ebc9b5b58923d9c37c/cores/nRF5/Adafruit_TinyUSB_Core/tinyusb/src/class/hid/hid_device.h)  There are a number of functions like tud_hid_keyboard_report

jpconstantineau, 2019-10-20:
for #1, we might be able to use  tud_hid_ready; again from tinyusb

JuanAntonioSEP, 2020-02-15:
Nice works on the port, looks promising for nextgen BLE 5 DIY keyboards. Was looking for a 65%ish keyboard, with hot-swappable switches and nrf52840 SoC (Usb and Ble). Would hit a sweet spot in my opinion. Low cherry Mx’s with suitably caps and some sort of slim power supply. Could be similar to the  SK621. Don’t know about the switch availability though. Anyhow. Is there a ongoing open source hardware branch like that, with embedded SoC, Pcb antenna, and stuff? If not, there should be. Think one could distribute the Pcb’s as development boards?

closed 2020-03-05 by jpconstantineau