jpconstantineau/BlueMicro_BLE #77: nrf52840 sending return keycodes

kohlerm, 2019-11-26:
I've got a feather nrf52840  express. I've got the firmware compiled and flashed after some trial and error. I flashed Iris default left firmware and I can connect to it and it shows up as a  keyboard. 
 But when I connect it sending return key codes at a very fast pace (I have not connected any switches) Is this expected?

jpconstantineau, 2019-11-26:
No It's not expected as-is.

What I suspect is happening is that you used a keyboard definition that
uses one of the GPIOs that's used for something else on the feather
nrf52840.

Have a look at the schematic for the feather here:
https://cdn-learn.adafruit.com/assets/assets/000/068/545/original/circuitpython_nRF52840_Schematic_REV-D.png?1546364754

Hopefully, it will guide you in selecting the right GPIOs for the rows and
columns.  The GPIOs are marked as P0.00 to P0.32 and P1.00 to P1.15.  Use
the number (add 32 for those on port 1)

Pierre

kohlerm, 2019-11-26:
Thanks!
That makes sense to me.

kohlerm, 2019-11-28:
Hmm, I replaced the pin numbers in some of the example boards, even cutting down the 25 board to 2x2. But I get still the same behavior. It actually confuses my Windows laptop so much that I have to reboot it. Any hints how to debug this? I'm a relative experienced developer, but I have no clue yet how to debug this.

jpconstantineau, 2019-11-29:
probably, the first step I would do is flash some simple adafruit example
from the arduino IDE and test it that way.

From there, I would use the embedded serial connection for debugging each
line to see if they map out fine.

kohlerm, 2019-11-29:
Simple adafruit examples work fine. I even tested some keyboard related examples. 
Ok will check whether I can find out something with embedded serial connection.

jpconstantineau, 2019-11-29:
If the adafruit examples are fine, then it's related to the firmware itself.

Just to be sure, what hardware do you have and which bootloader do you use?
There are a few different bootloaders for the different nrf52840 boards.
For the feather nrf52840, the feather bootloader should work fine. (there
is a separate flash memory chip if I remember well)
For the BlueMicro840 (nrfmicro variant), I have used the PCA10056 as the
reference and that seems to work fine with me - although I haven't tested
the firmware yet.

kohlerm, 2019-11-30:
Some progress. 
I modified the 25 keyboard reduced it to 2x2
using some pins which should not be used otherwise, see below.
It send they keys
1267  (makes sense given the layout) then repeats 7 infinitely. 

Is this expected?

In the test layout:
```
std::array<std::array<Key, MATRIX_COLS>, MATRIX_ROWS> matrix =
    {{
        {KC_1,    KC_2, },
        {KC_6,    KC_7,  }
     
    }};

```
```

#define MATRIX_ROWS 2
#define MATRIX_COLS 2

#define MATRIX_ROW_PINS { 26, 27 }
#define MATRIX_COL_PINS { 6,8 }
#define UNUSED_PINS {}

/* COL2ROW or ROW2COL */
#define DIODE_DIRECTION COL2ROW

#define KEYMAP( \
	 K00,   K01,       \
	 K10,   K11     \
) { \
	{ K00,   K01,   }, \
	{ K10,   K11,    } \
}

#endif /* KEYBOARD_CONFIG_H */

```

jpconstantineau, 2019-12-01:
I won’t be able to dive into the schematics for a while as I am travelling
for work.

By comparing the pin assignments, the columns look ok but I am suspicious
about the rows.

One element that hasn’t been accounted in the code is for gpios on port 1.
Unlike just reading single pins at a time, the whole port is read. However,
since pin assignment may span the two ports, I am not sure how it will work
with trying to get a bit beyond 32 bits of port 0, when in reality we must
read port 1.
In this case, everything seems to be on port 0; which would be fine.

Which branch of the code are you using?
If you look at the avr-mapping.h file in the media-keys branch, you will
see gpio to pin assignments for different hardware versions.

In this branch, I am converting all key code mapping to 32 bits; which is
still a work in progress. This will allow for media keys, macros and plenty
of other features we are looking for.

I’ll need to check on a few things when all my work travel is done.

kohlerm, 2019-12-04:
No hurry! I will only work slowly on this because of time constrains. 
Anyway I was working on master so far, I will check the branch you suggested. I haven't updated the firmware yet, therefore it should be the stock feather firmware.

jpconstantineau, 2019-12-05:
Time constraints... I understand. I am in between flights on business
travel...

I looked at the pins/gpios you used and I can't see anything wrong with it.
You avoided port 1 as I recommended.

When "7" gets repeated infinitely, this is probably because the chip
crashes and the computer doesn't receive an "all clear" from the chip. The
same happens if you type and walk outside of BLE range.

What I suspect may be happening too for the keypresses is that somehow the
chip uses internal pull-ups on the input pins or reads high when the pins
are tri-state. This will need further investigation.

Regardless if you use it plugged in a board or disconnected from anything,
the input pins shouldn't be high when not connected to any output pins
(through switches and diodes). Typically, the columns are pulled high,
current flows though diodes and are sinked to the rows, making them go
high. (there are variations but that's how I design my boards).  If the
feather is floating in mid-air and sends keypresses, we have an issue
somewhere. If the feather is connected to something, have a look if the
same thing happens when not connected to anything (except battery...)

I have a feather (both 832 and 840) and a Bluemicro840. I'll be able to
help test that in a few weeks.

jpconstantineau, 2019-12-05:
Re-opening.  This is an issue that needs to be resolved.
Support for the nrf52840 isn't there yet. It's not been tested yet, however, since the nrf52832 and nrf52840 are the same chip family, they should be able to support the same nrf52832 firmware with minor changes. (not accounting for the extra features that the 840 allows for)

Note that we will need to support 2 bootloaders: Adafruit Feather and the Nordic NRF52840DK

kohlerm, 2019-12-06:
It is now repeating 1267 endlessly. I'm not aware I changed anything. Maybe it is only Windows behaving differently.

jpconstantineau, 2019-12-06:
I won't be able to test it for another couple of weeks. In the meantime, we
can probably comment out or copy specific sections of the code to see where
these come from.

We should first focus on the scanning routine and the gpio read. By copying
the specific code used in the gpio setup and gpio read into an Arduino
sketch, we should be able to narrow it down and validate that it's coming
from the method for setting up the gpios.

Standard Arduino gpio reads may do something slightly different between the
832 and 840; this difference would need to be brought in for the Bluemicro
firmware. Unlike standard Arduino gpio reads, we read the entire port and
process it. (as opposed to reading each gpio one at a time).  This is where
I suspect we will need to make changes.

By reading the entire port, we avoid making repeated calls to the same gpio
read code but for different pins, making the keyboard more responsive and
lower power.

kohlerm, 2019-12-07:
if you could provide me some hints from where to copy the code, then i could do some tests

jpconstantineau, 2019-12-07:
Firmware_main.cpp
Function scanmatrix

The read is done in this single line:

pindata = NRF_GPIO->IN

There are some prep commands ahead of it and a loop to scan the pindata
after the read ..

jpconstantineau, 2019-12-09:
I just created a branch test-nrf52840 for you to try the bypassing the read.

Change is in the file and location I indicated previously (about line
120).
With this change, you should read absolutely nothing...

you can make changes to the code in that branch and we will be able to work
together to figure it out when I am back home...
working on 2AM jetlag...

jpconstantineau, 2019-12-09:
Next step will be to make a change to this line:
KeyScanner::scanMatrix((pindata>>(columns[i]))&1, millis(), j, i);

to this
KeyScanner::scanMatrix((digitalRead(columns[i]), millis(), j, i);

  (didn;t test...)  This will force a standard arduino read instead of a
full port read.
If this doesn't work (although much more slowly), then we will have to look
at getting the core in a simple arduino sketch.

kohlerm, 2019-12-13:
Some progress ..
I checked out the branch changed the code too
pindata = 0; // test reading nothing - this will check that the read behaves differently on the nrf52840...
    
    ```
 for (int i = 0; i < MATRIX_COLS; ++i) {
      KeyScanner::scanMatrix((digitalRead(columns[i])), millis(), j, i); 
      //KeyScanner::scanMatrix((pindata>>(columns[i]))&1, millis(), j, i);       // This function processes the logic values and does the debouncing
      pinMode(columns[i], INPUT);                                     //'disables' the column that just got looped thru
     }
```

When I now connect D6(?) to GND  I get the keys 16 (non-repeating). So it looks like this activates column 1. :-)
I cannot  get column 2 activated. 
I also do not understand why it is D6 which should be Pin 7 but I do not have Pin 7 in my matrix only 6 and 8.

jpconstantineau, 2019-12-13:
Sounds like we need to remap/test all the gpios from the feather.
Alternatively, we may want to check the library or boot loader code for the
mapping.

kohlerm, 2019-12-13:
The arduino IDE says Bootloader 0.2.13 Softdevice **S140** 6.1.1 .
Is that newer than expected maybe?

jpconstantineau, 2019-12-13:
S140 is for the nrf52840. 6.1.1 is the softdevice version. 0.2.13 is the
bootloader version from adafruit. I'll have to check which one is the
latest.

In any case, if you have the latest version of the board support package
("manage boards" in the Arduino IDE), you should have the latest released
version of the bootloader.

jpconstantineau, 2020-01-11:
How's the build?  Can I close this issue?

jturcotte, 2020-01-12:
I'm facing the same issue with the nrf5840. `NRF_GPIO->IN>>(columns[i])` yields 0 even then the contact is closed between the pins, but `digitalRead(columns[i])` behaves properly.

Where could be the culprit, maybe those `pinModeSense` settings get overwritten or something like that? It seems to only be called on wake up. I haven't have time to play around much with it, but maybe you have an idea about what could be the fix.

jpconstantineau, 2020-01-13:
I made a few changes in the media-keys branch. You will see that I have some compiler options to check if a nrf52840 is being used. Part of the new code performs 2 GPIO reads. One for port 0, the other for port 1. Then it goes on to scan the 32bits for the keys/pins to be read.   I tested that it compiles fine but I don't have my hardware quite ready yet. 

Not tested is the off by 32 calc needed for the pin numbers. 

I believe this code will eliminate the need for individual digital reads. 

I would really appreciate if you could let me know if the code runs as expected on both the lower port and upper port.

jturcotte, 2020-01-16:
I tested a bit and the media-keys branch didn't fix it for me but it was also not working when reading from 6, which is on port 0.

A bit of debugging and looking at the digitalRead code revealed that the issue is probably that `g_ADigitalPinMap` needs to be used to resolve the pin bits on the 840. The 832 has a 1-1 mapping written in this map so this would explain why it works without it there.

kohlerm, 2020-01-17:
I compiled the 25 board and flashed it using this branch. It does send 16 when I connect D6 to ground. 
Of course the pin names used in the keyboard_config.h are are not correct. 
  I might have to play with numbers instead of pin names there.

jturcotte, 2020-01-17:
I couldn't find a nice pinout diagram for the Bluefruit 840 like the one available for the 832, but this map in the code is actually the best documentation I could find for now:

```
const uint32_t g_ADigitalPinMap[] =
{
  // D0 .. D13
  25,  // D0  is P0.25 (UART TX)
  24,  // D1  is P0.24 (UART RX 
  10,  // D2  is P0.10 (NFC2)
  47,  // D3  is P1.15 (LED1)
  42,  // D4  is P1.10 (LED2)
  40,  // D5  is P1.08
   7,  // D6  is P0.07
  34,  // D7  is P1.02 (Button)
  16,  // D8  is P0.16 (NeoPixel)
  26,  // D9  is P0.26
  27,  // D10 is P0.27
   6,  // D11 is P0.06
   8,  // D12 is P0.08
  41,  // D13 is P1.09

  // D14 .. D21 (aka A0 .. A7)
   4,  // D14 is P0.04 (A0)
   5,  // D15 is P0.05 (A1)
  30,  // D16 is P0.30 (A2)
  28,  // D17 is P0.28 (A3)
   2,  // D18 is P0.02 (A4)
   3,  // D19 is P0.03 (A5)
  29,  // D20 is P0.29 (A6, Battery)
  31,  // D21 is P0.31 (A7, ARef)

  // D22 .. D23 (aka I2C pins)
  12,  // D22 is P0.12 (SDA)
  11,  // D23 is P0.11 (SCL)

  // D24 .. D26 (aka SPI pins)
  15,  // D24 is P0.15 (SPI MISO)
  13,  // D25 is P0.13 (SPI MOSI)
  14,  // D26 is P0.14 (SPI SCK )

  // QSPI pins (not exposed via any header / test point)
  19,  // D27 is P0.19 (QSPI CLK)
  20,  // D28 is P0.20 (QSPI CS)
  17,  // D29 is P0.17 (QSPI Data 0)
  22,  // D30 is P0.22 (QSPI Data 1)
  23,  // D31 is P0.23 (QSPI Data 2)
  21,  // D32 is P0.21 (QSPI Data 3)

  // The remaining NFC pin
   9,  // D33 is P0.09 (NFC1, exposed only via test point on bottom of board)

  // Thus, there are 34 defined pins

  // The remaining pins are not usable:
  //
  //
  // The following pins were never listed as they were considered unusable
  //  0,      // P0.00 is XL1   (attached to 32.768kHz crystal)
  //  1,      // P0.01 is XL2   (attached to 32.768kHz crystal)
  // 18,      // P0.18 is RESET (attached to switch)
  // 32,      // P1.00 is SWO   (attached to debug header)
  // 
  // The remaining pins are not connected (per schematic)
  // 33,      // P1.01 is not connected per schematic
  // 35,      // P1.03 is not connected per schematic
  // 36,      // P1.04 is not connected per schematic
  // 37,      // P1.05 is not connected per schematic
  // 38,      // P1.06 is not connected per schematic
  // 39,      // P1.07 is not connected per schematic
  // 43,      // P1.11 is not connected per schematic
  // 44,      // P1.12 is not connected per schematic
  // 45,      // P1.13 is not connected per schematic
  // 46,      // P1.14 is not connected per schematic
};
```

So to use the pin labeled "6" on the board (D6), one should use `6` in `MATRIX_ROW_PINS` and `MATRIX_COL_PINS`. Not understanding the code I first got confused by the `D6` macro of avr_mapping.h used in some of the keyboard configs.

kohlerm, 2020-01-17:
https://learn.adafruit.com/assets/68545 ? < But yes that source code file is the best I could find as well.

jpconstantineau, 2020-01-17:
This is the equivalent mapping for the feather 840. The index in the matrix
is the digital PIN number and the number in the matrix is the gpio on the
chip itself.

What hardware do you have? A bluemicro or a feather?

jturcotte, 2020-01-17:
I meant for example this for the 832 that lists the IDE pin numbers, which is what should appear in the keyboard matrix definition: https://learn.adafruit.com/assets/46248.

I'm using a feather 840.

jpconstantineau, 2020-01-17:
ah... that explains things...

The firmware reads directly from the GPIO and uses the GPIO number to pick
up the value. not the pin number/label from the feather or bluemicro.

the BlueMicro firmware is intended to work with the BlueMicro hardware.
The BlueMicro hardware shapes a feather into the pinout of an arduino
promicro.  The AVR file essentially maps the ports that are on the arduino
pro micro to the nrf52 GPIO.  This is to simplify the work of people simply
copying their QMK configuration to the BlueMicro firmware. (It's a simple
copy/paste)

I think i will add entries for the feathers to the avr file so that  the
mapping between digital pins D0-D32 and the GPIO is done -
something similar to the function you indicate.

If you try using the gpio numbers instead of the D?? values, (the mapping
from that structure) you should be fine.

At some point, I did find the diagram for the feather nrf52840 and saved it
for reference:
https://github.com/jpconstantineau/NRF52-Board/blob/master/Reference/circuitpython_nRF52840_Schematic_REV-D.png

jturcotte, 2020-01-17:
The gpio numbers don't work well because the port->IN register expects port bit numbers, but digitalRead and digitalWrite wants indices in that mapping structure (IDE "D" pin numbers). I don't mind using either, but right now the rows are using digitalWrite and columns are reading directly from the IN port register, so it's not straightforward to maintain a matrix config that uses port indices (+ 32 for port #1) for columns but pin numbers for the rows.

I understand that the BlueMicro should be first citizen in the code and build instructions. The feather is however a very nice board too IMO for hand wire builds, and it would be nice if it was supported on the side. The BlueMicro firmware is probably the best keyboard firmware to use with it at the moment, given that QMK support is unlikely for license reasons and probably less battery-life optimized according to what I read (system off support, etc.).

That diagram is very nice, but I couldn't figure out from it how e.g. the A1-A5 pin numbers map to the seemingly random digital pin numbers Adafruit assigned to them in the software stack.

jpconstantineau, 2020-01-17:
I want to be able to support both feathers as well as the Bluemicro (in all
the version I have so far).

if we were to use for the row/col definition labels like D0, D1, D2... D?
we would be able to make it work?
thinking of creating a section in the hardware and avr file for a feather
definition with all the defines/mapping needed for D* GPIO*...
All you would need at the top of your keyboard_config.h file is:
#include "hardware_variants.h"
#define HARDWARE_MAPPING  FEATHERNRF840
#include "avr_mapping.h"

and then use

#define MATRIX_ROW_PINS {D0,D1, D2, D3 } // set the actual pins you use -
as adafruit labelled them
#define MATRIX_COL_PINS {D4, D5, D6, D7, D8, D9, D11 } // set the actual
pins you use - as adafruit labelled them

Do you think that could work out well?

jturcotte, 2020-01-17:
For my case, or for hand wire builds in general on a feather (assuming few widely distributed PCBs will support feathers) I think that having a local branch and adjusting the matrix according to how things got soldered is perfectly acceptable. So `HARDWARE_MAPPING` would be fine to me.

What I find weird is to piggy-back an nrf52840 BlueMicro as a feather board variant in the Arduino IDE, an abstraction layer whose purpose is already to handle things like pin-to-port mapping, and then add an extra layer on top to map ProMicro pin numbers onto it.
I don't know if it's feasible, but I think the cleanest solution would be to find a way to have the BlueMicro directly in the Board Manager and list of boards of the Arduino IDE to be selected before the build, and encode the existing avr_mapping.h file instead as a g_ADigitalPinMap in its own hardware variant.cpp file.

But I don't know how easy it is to have this defined outside the Adafruit Arduino package while still depending on it, directly in the firmware source tree, or as a separate Board Manager URL in the setting.

kohlerm, 2020-01-17:
some success here!
I've put
#define MATRIX_ROW_PINS { 26, 27 ,  6, 8, 4 }
#define MATRIX_COL_PINS { 5, 2, 3, 10, 11 }

into the 25 board. 
Connecting A1 with ground outputs afkp, which is the first column :-)
you have to build with
.\build.ps1 25:test:master -nrf52840

wasn't aware of that new  option.

jpconstantineau, 2020-01-17:
Getting the board in the boards manager would be great. One of the problem
associated with that is that I don’t use the arduino ide for coding or
flashing. I have automated builds that call the arduino-cli in docker
containers. These build every keyboard/key map for both the 832 and 840.
There are no hardware tests but I know pretty quickly if a core change
breaks something somewhere.

The boards also come with their own boot loader; something I really don’t
have time to customize. That’s why we piggy-back the feathers.

It would be possible to create boards in the ide and re-configure all the
builds. It would take time: something I am short on.

Regarding the digital read. We want to read all pins all at once instead of
reading each pin, one at a time (what using the standard feather digital
read would do). This ensures a faster cycle time between each scan; and a
lower power consumption since it would go to sleep faster. (Not quite a
sleep but the CPU wouldn’t run as much)

jturcotte, 2020-01-17:
OK I understand, I don't think doing this super cleanly is necessary either, it just needs to be somewhat reliable.

About using `NRF_GPIO->IN` I think it's right to avoid re-reading that register for each column, I just think that we shouldn't mask using values from the matrix definition directly unless we also stop using `digitalWrite` for writing to rows. Those values would need to be passed through `g_ADigitalPinMap` (or the convenience macro `digitalPinToBitMask`) to be consistent with the digital pin numbers that `digitalWrite` takes as input.

A solution could be to map digital pin numbers to port pins before they are put in the `row` and `column` keyboard matrix variables, or later each time they are tested against bits of `NRF_GPIO->IN`.

I'm still in the process of assembling everything, but I can have a look at getting the feather 840 code in shape once I have everything running.

jpconstantineau, 2020-01-17:
I see what you mean... the digital write and pin modes are being used and
there is very likely a bug where the GPIO is entered when it's the pin
that's expected...
The same would apply to the sleep functions when the pins "sense" mode is
turned on.

kohlerm, 2020-01-18:
I did some more tests and it seems to work fine for port0 pins, both for columns and rows. 
I tried D13/41 as a row port and connecting to a column pin at A1/5, which did not work. 
Is this expected at the current stage?
My main goal at the moment is to figure out how may pins I can use ...

jpconstantineau, 2020-01-19:
Hi Guys,

I made a few  changes to the media-keys branch.
1 - copied the wiring-digital files from the adafruit libraries and renamed all the functions to "GPIO" functions.  This way, we bypass the g_ADigitalPinMap "function" to map pins to GPIOs.  By calling the new functions, we expect the GPIOs.   For GPIOs in port 1, add "32" to the bit number. I tested them with a simple blink sketch. 
2 - changed the firmware_main functions to call the new functions.  As such, we no longer call the default adafruit/arduino pinmode, digitalread and digitalwrite functions.

I suspect that the battery service - which calls the analogread function - will need updating.

jpconstantineau, 2020-01-19:
I did some more checking about the mapping done in the variants. What I see is that the nrf52832 feather goes 1-1 between pins and gpios. The nrf52840 feather mixes things up. The nrf52840 dk (pca10056) also does 1-1. When I flash my test boards, I use the pca10056. For the nrf52840 feather, I think using the mapping function will make more sense than removing them all since it breaks all the other libraries that use digitalread and digitalwrite.   Some more thinking will be needed when defining the rows/columns and make sure that it all works with the feather.

jturcotte, 2020-02-09:
For the record, I've tried the media-keys branch with the nrf52840 feather. While I don't have any column on port 1 so I only tested port 0, I think that the code looks good.

In retrospect though, I wonder if avoiding the read of the ports on each pin really does save a lot of battery. From what I've read, reading the GPIO just involves reading an IO register and this can't be that slow, can it? So I'm wondering if it would be bad to just call `digitalReadGPIO` each time to avoid having two implementations.

jpconstantineau, 2020-02-09:
I wonder as well. Going from memory of the spec, Battery usage is highest
when the radio is active (15-20mA), followed by when the cpu is
active(8mA). It’s lowest when everything  is idle (1-3uA range).

I wonder if 1 call to the register and looping over the Pin list and
getting the bit from the 32 bit integer takes less time/instructions over
looping over the pin list and calling the standard read, which itself reads
the 32 bits register and gets the bit out.

A small arduino sketch could be created to experiment/test the two. I have
a power profiler from Nordic that would reveal the difference.

I have a folder with arduino tests. If you want, you can create 2 sketches
in there and I can run the profiler on them.

If the difference is trivial, I would be glad to standardize on the normal
read.  I would use one of my old lipo-battery-less designs and connect the
profiler directly to the blue micro v1.x on a contra (4x12) to test it out.
This eliminates the influence of any external components.

I have tested the nrd52840 code on a board that uses  a single upper
register pin and it worked fine. I will be bringing the nef52840 changes to
the develop branch but I have to extract them from the other stuff in the
media branch. The media branch also went from 8bit key codes to 32 bits key
codes and that made simpler the implementation of media keys and macros (by
using specific bits to flag the other functions) however, I am running into
stability issues when lots of layers are present.  Extracting the 8bit
keycodes out for every key seemed to also slow down the whole thing and I
can’t type too fast or keys pressed get missed.

Of note: adafruit just released 0.18.0 which brings the functionality of
key sense to their standard library. As such, the sleep functionality will
be able to be standardized on the library. With that, I will drop all the
“gpio” pin functions and use the standard ones.

jpconstantineau, 2020-03-05:
This has been resolved and brought in develop

closed 2020-03-05 by jpconstantineau