jpconstantineau/BlueMicro_BLE #82: The input is sluggish, typing fast can lead to lost keys

jturcotte, 2020-02-09:
**Describe the bug**
I see the issue the most when doing Alt-Tab multiple times, I'm used to type exactly a certain amount of keys repeatedly very fast and I was wondering if this was a bug in Gnome or something.

I happened to see that the keyboard reports are sent for each scan when any key is held, instead of only when the report would change. I found this a waste of battery, and trying to [hack a quick solution](https://github.com/jturcotte/megatreus-firmware/commit/3fdc0ae5425167a47f22d6d138421075be3e0b44) I realized that it actually solved the issue for me!

The problem is that when trying to rebase it on top of the media-keys branch to make sure the code won't need to be merged back to develop, I noticed that something else makes the reports **very** sluggish in that branch, even with my hack. I tried to investigate but couldn't find why it was so slow. Maybe you have an idea.

**To Reproduce**
Steps to reproduce the behavior:
Type fast

**Expected behavior**
The keyboard feels as responsive as a USB keyboard.

**Desktop (please complete the following information):**
Only tried on Linux

jpconstantineau, 2020-02-09:
Great! Can you bring your solution into a branch Of develop?  I will be
bringing the nrf52840 changes from the media-keys branch out into its own
branch so that we can test if your solution will work out too.

There are too many things in the media-keys branch that testing each
feature individually will be needed to weed out the issues in it.

jpconstantineau, 2020-03-05:
This has been brought in develop

closed 2020-03-05 by jpconstantineau