jpconstantineau/BlueMicro_BLE #88: USB HID for NRF52840

jpconstantineau, 2020-03-05:
Use TinyUSB that's part of the adafruit library to detect when we need to csend keypresses through USB instead of BLE

Zygriid, 2020-04-06:
Hi!
Have you had any progress on this?
I am intersted in nkro for stenography. 
How could i help?

jpconstantineau, 2020-04-06:
Hi

I started looking into it but haven't made much progress. One needs to use
the tinyusb library to get the USB hid pieces in place.  We would need a
basic example where we could switch from USB to BLE easily so that we can
port it to the firmware.

For nkro, I assume we would need to do something similar to what other
firmware do for it. I honestly don't know what's needed to do it. Standard
USB and BLE HID has buffers for 6 keys plus the modifiers.

As I indicated, I haven't put too much time on it yet as I have started
looking into bringing support for split boards to all the changes and new
features I implemented in the branch I have been working on.

I do have a plover layout for the contra. For steno, this was the easy
part. Nkro is the next and final part needed for it.

jpconstantineau, 2020-05-13:
I got something started for USB HID for the nrf52840.  We now need to include the tinyusb library and bring this example: https://github.com/jpconstantineau/BlueMicro_BLE/tree/master/arduino_tests/nrf52840_hid_keyboard
in the main code.

akudaikon, 2020-09-05:
Hey, just as an FYI - A month or so ago I forked off the BlueMicro firmware and made quite a few changes, like USB support (with graceful handoff between USB and BT), lock LED support, and more. While some of the changes are lazily implemented and very specific to the keyboards I wanted to use it with (an IBM Model M and Model M2 with custom hardware based off an Adafruit NRF52840 Express), feel free to pull over any changes you'd like, if it'd help and you find them useful!

[https://bitbucket.org/akudaikon/bluemicro-model-m/src/master/](https://bitbucket.org/akudaikon/bluemicro-model-m/src/master/)

mpaupulaire4, 2020-12-27:
Hi has there been any progress on this?

jpconstantineau, 2020-12-27:
I haven't had a chance to  work on this and will need help doing so as I time is pretty limited. one could take akudaikon's code and see how to merge it within the overall Bluemicro firmware.

jpconstantineau, 2020-12-31:
@akudaikon Jason: I have been trying to get your fork merged into my code for USB and profile support and noted that you specifically call for the function getediv, which isn't defined in the adafruit library.
Which version of the library are you using? If you have modified yours, perhaps you can bring the changes to the community BSP so that we can get everything in.

jpconstantineau, 2020-12-31:
@akudaikon I just added the function to the [community BSP](https://github.com/jpconstantineau/Community_nRF52_Arduino).

akudaikon, 2020-12-31:
@jpconstantineau Merging all of my changes might not be the best approach because I've modified the code to specifically suit my needs, and most of those changes probably won't be useful for BLE Micro.

I added the getEdiv function, like you assumed. It was a quick workaround to get the profiles to work how I wanted (specifically for unpairing a certain profile). I believe the newer versions of the Adafruit BSP get rid of the ediv entirely and use a different form of identification, so be aware of that.

jpconstantineau, 2020-12-31:
@akudaikon I am not planning to make bulk merge in but cherry-pick specific features one at a time.

Some of the changes they have done to use the cryptocell instead of the previous method for handling pairing caused significant power consumption (see [issue 600](https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/600))  That's why the community BSP took 0.21.0 as the starting point as opposed to their head.  I suspect syncing the repos will be a pain but that cryptocell power consumption was a deal-breaker...

So that you are aware, using PWM (analog write) can use quite a bit of power (more than one would expect). See [here](https://devzone.nordicsemi.com/f/nordic-q-a/40912/pwm-power-consumption-nrf52840)  That's why I don't use analogwrite but go directly to the peripheral instead and turn it off when not in use.  I'll have to investigate if they also turn it off when not in use but at first test, they didn't.

jpconstantineau, 2021-01-01:
Hi all, keep an eye on PR #187 It contains the contributions from @akudaiko.  Work is still in progress to merge the functionality in and very likely some refactoring will be needed to simplify/deduplicate some of the code but this will be a start.

Since we need TinyUSB library and that it's included in the Community BSP, I will be requiring to use the Community BSP for enabling this functionality (the Community BSP is needed for other BLE functionality too).  As such, you will see in the code specific compilation-time definitions to see if we are using one BSP over the other.  As such, if you compile with the Adafruit BSP, it will compile fine but USB functionality won't be available.

jpconstantineau, 2021-01-24:
PR #187 brought in the changes into develop. Closing.

closed 2021-01-24 by jpconstantineau