jpconstantineau/BlueMicro_BLE #111: Installation instructions and guides

bli00, 2020-05-20:
I'm really looking forward to implementing this on my new keyboard build. But it seems like the documentations are a bit lacking in terms of instructions. Perhaps there are common knowledge about keyboard building I should familiarize myself with first but it'd be great if you guys can provide some basic instructions on how to install this on a pcb (and/or a breakout board).

jpconstantineau, 2020-05-20:
I agree the instructions are a bit thin.  I am setting up a tutorial on my
youtube channel.  It's still ongoing but your timing is good as I am
setting up a playlist over the next few days:
https://www.youtube.com/playlist?list=PL7thPYHHuORO1QoXQUBm_7NkGXwrC4qEH

Are you looking specifically for setting up the hardware or the firmware?

bli00, 2020-05-20:
@jpconstantineau My opinion would be to add something that caters more to the general audience. Personally, I'm looking for instructions how to get a bluetooth keyboard running. So any of the following categories would help:

* What electronic parts are needed.
* An explanation of the parts and what to look for when buying.
* An ELI5 explanation of how BlueMicro works.
* What's **not** supported.
* What needs to be soldered in terms of the breakout board.
* What's needed to connect to the system start setting up.
* How to flash the firmware.
* How to implement a layout.
* How to troubleshoot and some common issues.

I guess these bullets points are a bit abstract and vague. Personally, I have never built keyboard before so I welcome basic tutorials. It's up to you how advanced you want the documentations to be.

bli00, 2020-05-20:
The following juxtaposition contrasts two different documentation complexities/styles (Of course these aren't actual instructions and I'm not sure how it works myself, or else I won't be here XD):

> BlueMicro is a firmware that instructs a PCB (the circuit board underneath your keycaps) with bluetooth capabilities to communicate with your computer. A firmware is a piece of software that's installed onto the PCB like an operating system. Here is a list of bluetooth PCBs that's supported. To flash (or install) BlueMicro onto your PCB, you will need the following software installed...

> Import the ___ library under ___ directory. Create a file named ___.cpp and copy and paste [this] template. Change the template according to your layout. Documentation for they key constants is [here]. You'll need to build with GCC Versions 999 or greater...

I think the first one is always preferable but more time consuming. I'm looking for someone to hold my hands for my first build but you might not deem that necessary for your targeted audience. Either way, more documentations are always appreciated.

jpconstantineau, 2020-05-20:
I see...  This repo is firmware specific. There are multiple keyboards that
it supports and going through every one would be too complex.

How to build a keyboard/what parts to look for: probably starting with the
wired instructions would be a good start. There are multiple keyboard
builds out there, including the one I designed a few years ago.
See its build guide here:
https://github.com/jpconstantineau/ErgoTravel/blob/master/BuildInstructions.md

The firmware explicitly targets the nrf52832 and nrf52840 chips.  Any other
processors won't work.  The Adafruit Feathers are supported (this is one
type of breakout boards) and the breakout boards that connect to a keyboard
PCB through the Pro Micro footprint are supported.

As for the later steps (layout, build, flash), I will be covering through
my tutorial series prior to documenting in a web page and
troubleshooting/common issues, I think building up a FAQ would be useful.

Orphan-Crippler, 2020-08-21:
Is there a clearer pinout diagram? Or at least one that correlates to what i see people are putting in there keyboard_config.h, looking through all the ones posted i see some people are using only numbers and others are using letter number combos like QMK with pro micro. I have been looking at the pinout on nice!nano website for about an hour now and cant seem to figure it out.

BenRoe, 2020-10-19:
@Orphan-Crippler if i understand it correct

- Pin P0.02 (002 silkscreen on nice!nano) is pin 2
- Pin P1.11 (111 silkscreen on nice!nano) is pin 43 in
`firmware/keyboards/crkbd/hardware/pca10056/nicenano/hardware_config.h` 

@jpconstantineau explained it here https://github.com/jpconstantineau/BlueMicro_BLE/issues/155#issuecomment-696452471

> For all other boards (PCA10056), use this formula:
> 32*X + Y
> where the GPIO is described as PX.Y
> 0.11 = 11
> 1.11 = 32+11 = 43
>

jpconstantineau, 2020-10-19:
I am thinking of adding a macro you can call.  Something similar to:

#define _PINNUM(port, pin)    ((port)*32 + (pin))

It's already being used for some of the adafruit boards:

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/eefeba8d64dfdae15a5469447664eea16d32816e/variants/particle_xenon/variant.h#L22

BenRoe, 2020-10-19:
> I am thinking of adding a macro you can call. Something similar to:
> 
> #define _PINNUM(port, pin) ((port)*32 + (pin))

If that means we can use the following, then it would be easier.
```
#define MATRIX_ROW_PINS { 0.13, 0.27, 0.04, 1.43 }
```

jpconstantineau, 2020-10-19:
No, it would be:
#define MATRIX_ROW_PINS { _PINNUM(0,13), _PINNUM(0,27), _PINNUM(0,4), _PINNUM(1,43) }

BenRoe, 2020-10-19:
Not sure. It looks more complex/unclear.
Easy implementation would be a simple comment where pin numbers are used.
After i have read `1.11 = 32+11 = 43`, it was clear.

jpconstantineau, 2020-10-25:
@Orphan-Crippler Have a look at the new documentation page and let me know if that helps:
https://github.com/jpconstantineau/BlueMicro_BLE/blob/develop/docs/configuring_firmware.md

Orphan-Crippler, 2020-11-07:
@jpconstantineau That is just fantastic, Thank You so much!!!!

closed 2021-11-28 by jpconstantineau