jpconstantineau/BlueMicro_BLE #135: BLE Profiles for connecting to multiple computers

jpconstantineau, 2020-07-29:
**Is your feature request related to a problem? Please describe.**
When using a keyboard with multiple computers (desktop and laptop for example), I want to be able to quickly switch the connection from one to the other without having to unpair/forget the keyboard on the computer.

**Describe the solution you'd like**
The firmware should allow for multiple "Profiles" where different connections are available.  The profile could be toggled from one to another using a specific keyboard macro/key press/sequence

**Additional context**
This would require testing using a simple adafruit example and potentially extending the adafruit BSP to support this type of functionality.

jpconstantineau, 2020-09-01:
Work has started on this: https://github.com/jpconstantineau/BlueMicro_BLE/tree/BLE-Profiles

mpaupulaire4, 2020-12-27:
Has there been any more progress on this? How can I help support?

jpconstantineau, 2020-12-27:
I haven't made progress on this recently.  To help, we need to take a simple example, and use the setAddr and getAddr functions of the Bluefruit library to change the BLE address of the keyboard so that a computer that's connected to a specific BLE address would not reconnect automatically and have another computer/device allow to connect to the new address.

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/src/bluefruit.cpp#L517

The simple keyboard example from the adafruit library would be a good prototype to get this developed and tested; without all the overhead of a complete keyboard firmware.  Once proven and tested, we can migrate the code to the Bluemicro firmware.

jpconstantineau, 2020-12-31:
I just included the changes by @akudaikon and tested them on my luddite.  They work great!  Switched between by phone and computer without any issues. 

Note that you will need to switch over to the Community BSP in order to enable this.  This is very simple to do - I'll update the scripts and the docs.  In the meantime, watch for PR #186 and get ready by installing what's needed: [Community BSP](https://github.com/jpconstantineau/Community_nRF52_Arduino/wiki/Installation-Instructions)

akudaikon, 2020-12-31:
Glad to hear it! Sorry my changes aren't the easiest to pull in. I was initially intending to just modify BlueMicro for my own needs (and therefore some of the changes are implemented pretty lazily) until I saw some of the feature requests for things I already implemented on here. I'm glad they're helpful though!

jpconstantineau, 2021-01-24:
PR #186 has been brought into develop.  Closing

closed 2021-01-24 by jpconstantineau