jpconstantineau/BlueMicro_BLE #152: Combos support?

arturbecker, 2020-08-30:
Couldn't find it on the comparison page in the docs... does BlueMicro support/plans to support [Combos](https://beta.docs.qmk.fm/using-qmk/software-features/feature_combo)?

jpconstantineau, 2020-09-01:
Looking at the config from QMK, it looks like it would be possible to implement similar functionality relatively similarly.  It would be configured quite differently.

Are there specific examples of key replacement you would want to have considered?  

I would add a hook between the keymap processing and sending of the keys via BLE so that key replacement could be processed.

arturbecker, 2020-09-01:
Thanks for the reply! What do you mean by "replacement" in this context?
If it is helpful, here's one of my use cases:
I map the combo of `s`+`d` to `(`, and the combo of `k`+`l` to `)`. It's like space cadet shift, but with combos on home row keys instead, if you're familiar with it. I also map the combos on the reverse order (i.e. `d`+`s`) to the same key, to avoid missing activations. Then, my combo timeout is set to 17ms.
Other than basic keycodes, in QMK it can also be used to activate layers. I used it to momentarily activate a layer too, but now I mostly only send basic keycodes and shifted keycodes such as `{` and `}`

jpconstantineau, 2020-09-02:
I was thinking replacement as in:
if both "s" and "d" are in the buffer to be sent, delete these and send "(" instead.  Sounds simple...  Simple to implement...

This to be processed for every scan but as the last step - just before sending keypresses out. This does have the potential problem that sometimes, the initial key might be sent even though on the next scan you have both.  And on release, you may have the "lagging" key being sent. (lagging - the one unpressed last). I do see the potential need for a timeout on both press and depress.  Implementation with timestamps will not be as trivial...
Because it would be processed last, it wouldn't be able to switch layers (momentarily or otherwise) on that scan.  I think we would need another "activation" layer or "special key modifiers" to differentiate keys that trigger sequences; not sure what exactly. 

Also, you indicate that it picks up a sequence of keys, not simultaneous keys...  
You had to create 2 sequences to map it properly. "sd" and "ds"; both to be replaced with "("

Do you see the need for the two sequences to be able to be differentiated?

arturbecker, 2020-09-02:
Thanks, I think I understand it now.
For my use case it's honestly better to NOT have sequence matter (contrary to how it is on QMK). That's because I only use combos with short timeouts. If you use medium timeouts, maybe it's possible to consistently type out inner (`s`+`d`) or outer (`D`+`S`) sequences at will; but then that might start to interfere with normal typing if your typing speed is high.

The other use case for sequences mattering that I see is if you want to define combos for e.g. [very uncommon/non-existent bigrams](https://norvig.com/mayzner.html), such as `j`+`q`. So you could press `j`, and have an insanely long timeout to press `q` (which wouldn't bother the regular typing experience as long as you don't want to send `j` by itself... then that's an issue). But I never saw such use case in the wild. It would more or less simulate a Leader key with Combos, if that makes sense.

P.S.: I'm not really familiar with QMK's code, so I can be using the wrong terminology... from what I understand, in my `s`+`d` example, what happens is:
I type `s`; then:
if I type any other key but `d`, send both s and this other key
if I press `d`, send `(`
if I take more than 17ms to type anything else, send `s`

jpconstantineau, 2020-09-02:
What about modifiers?
Modifiers are treated separately in the HID standard.  There is a buffer for standard keys and there is a single byte bit-masked for modifiers.
Take for example when looking only at the keycode buffer:
`s` + `d` is equivalent to `S` + `D`.  The only difference is that the bit for one of the shift is set in the modifier byte.

Do you think it would be OK not to account for the state of the modifier. Or would any modifier essentially cancel the combos?

Finally, I think it would be still be possible to get timing information as part of the post-processing logic. Perhaps that's where Combos and Sequences differ...

```

Combo (FIND_KEY1, FIND_KEY2, REPLACE_WITH_KEY3)             //  keys must be pressed in a single scan
Sequence (FIND_KEY1, FIND_KEY2, REPLACE_WITH_KEY3,SEQUENCE_TIMEOUT)  //  keys must be pressed on consecutive scans

```

I potentially see the need for longer sequences (think vi commands... `:wq` for example)
I also see this as potentially triggering macros or special keyboard functions.

tjleft, 2020-09-14:
Hello! I am also interested in the implementation of combos. Modifiers should not cancel the combos but should carry through to the sent key.

jpconstantineau, 2020-09-14:
For anyone interested in combos, I need help to get this implemented.

Here are a few ideas on how to start a prototype:
1 - Clone the repo and create a "combo" branch
2 - In Keyscanner.cpp, in the function KeyScanner::getReport(), implement a call to a new function called "process_combos"
3 - Similarly to how macros are processed (see process_user_macros in firmware_main.cpp), create a if define statement to check if a USER_PROCESS_COMBOS exists
3 - create an empty process_combos wrapped with the above USER_PROCESS_COMBOS check.  This function should do "nothing" as there is nothing to process by default
4 - in keymap.h, define the USER_PROCESS_COMBOS =1
5 - in keymap.cpp create process_combos function to process the combos.
6 - process the KeyScanner::currentReport[8] and intercept the combo combinations you need to process... 
7 - the function should remove any intercept and add/swap any replacements. 

currentReport[0] are the modifiers in bitmasked format
currentReport[1-6] are the keycodes being sent.  These are AFTER layers processing.  

Hopefully this will give enough to help someone develop a prototype of this feature.  This will likely apply to both combos and leader keys.

arturbecker, 2020-09-15:
Sorry I missed your reply Pierre... I personally don't use modifiers with my combos at the moment, but I agree it would probably be desirable to have them "carried through" to the sent key.
As for Leader keys, I thought they were already implemented?

tomsaleeba, 2021-01-28:
My [fork](https://github.com/jpconstantineau/BlueMicro_BLE/compare/develop...tomsaleeba:combos) is working in that you can perform the combos, yay!

However, it needs some debouncing. If you flash the [combos](https://github.com/tomsaleeba/BlueMicro_BLE/tree/combos/firmware/keyboards/4x4Tutorials/keymaps/combos) keymap and press `1` and `2` to trigger the combo for `a`, you'll get the `a` typed but most times you press it you'll also get a `1` and/or a `2` also printed. I'm pretty sure this could be fixed by adding a debounce where we wait for the pressed keys to settle down, before we look up the combo for it. Right now I think the fast probing of the firmware along with the fact the key switches don't immediately go from open to closed means the firmware will *see* one switch pressed, then both, then maybe one again all in a very short amount of time.

Right now, I don't know where to add the debouncing. Is it something specific to combos or is it keyboard-wide. I can see there's already debounce logic [here](https://github.com/jpconstantineau/BlueMicro_BLE/blob/7e7d644/firmware/KeyScanner.cpp#L39) and [here](https://github.com/jpconstantineau/BlueMicro_BLE/blob/6764cc3/firmware/firmware_main.cpp#L276).

My next step is to play with `DEBOUNCETIME` values and experiment with debounce logic just for combos. If anyone has any tips, I'll happily accept them because I'm in a bit over my head here.

jpconstantineau, 2021-01-28:
Hi Tom,  We need to look carefully at both debounce logic and see which one really applies as of right now.  There has been some changes a  while ago and I don't think we removed the dead code (if it actually is indeed dead)  I believe it's the one in firmware_main that's being used.  you can try commenting out the one in keyscanner and see if compilation succeeds or fails.

From looking at your process_combos function, it seems that it would get called at every scan (or anytime there are keycodes in the buffer).  As you indicated, it's very likely that one key will be pressed, then the other, then the final one; each being processed during different scans.

I suspect that the process_combo function will need to remember the trigger keycode when pressed, remove it from the buffer, and that if it doesn't detect the second keycode within a certain amount of time in thee buffer (next scan potentially), then it returns the initial keycode in the buffer to be sent.

For doing so, you will need static variables within the function as well as check the "millis" function to see at what time we got the press.  Alternately, we could see if we could add time to what gets passed to the function (perhaps even the entire keyboard state structure if that's needed and simpler)

I also suspect that with USB connection, we have some funny things happening  with either the debouncing logic or the timing when reports are being sent as I sometimes have missed reports (missed keys or missed "all clear" reports).  I'll turn off battery monitoring to see if calling battery ADC when typing would be the cause or if it's unrelated.

jpconstantineau, 2021-01-28:
Note that the order of the keys in the buffer depends on the sequence in the matrix and not when they are pressed.  As such, if 2 is scanned before 1 (in the nested for loops), it will show up first.

As such, combos should look for keys in any of the 6 positions in the buffer and not assume that they will be at a specific location in the buffer.

tomsaleeba, 2021-01-28:
> Note that the order of the keys in the buffer depends on the sequence in the matrix and not when they are pressed. As such, if 2 is scanned before 1 (in the nested for loops), it will show up [...]

Thanks for the tip :+1: I'll make that change.

jpconstantineau, 2021-01-29:
I just tested removing the scanmatrix from the keyscanner class.  Still compiles fine. I'll remove and test further but I think we will be good removing that dead code.

jpconstantineau, 2021-01-29:
I just pushed a new branch with the debounce dead code removed and rescheduled battery monitoring.  I also brought in the keyboard dynamic state within the keyscanner class so that you could refer to the timestamp in it.

My luddite seems to work fine with these changes but I'll wait for a bit longer term usage to see if the issue with missed reports/debounce is still present.

jpconstantineau, 2021-02-20:
@tomsaleeba  Now that I have tested Combos on a QMK board (Curtesy of @germ here in town), I really get what these are about and have a good use case for them.

Have you made progress in your fork on them?

tomsaleeba, 2021-02-20:
@jpconstantineau I make heavy use of combos in QMK as you can see from my [userspace code](https://github.com/tomsaleeba/qmk_firmware/blob/70fcff8/users/tomsaleeba/tom.c#L220).

I want combos in BlueMicro to play with the https://www.artsey.io/ layout. The key requirement here is that I need *only* the longest combo to be triggered as there will be overlapping macros.

But no, I haven't made any more progress. I've played with your latest `timestamp` and `latestupdatetime` additions but haven't had a go at the logic to "debounce". I'm not a C programmer so I can't smash out the code for this as quick as I'd like.

jpconstantineau, 2021-02-20:
@tomsaleeba  [this layout?](https://www.artsey.io/_FILES/images/ARTSEY.jpg)  If I make this work, it will be able to handle pretty much anything...  

I am thinking of supporting @germ combo engine configuration [here](http://combos.gboards.ca/docs/combos/).  The COMB() macro is specifically what I am looking at.  The SUBS() Macro could leverage the existing macro functionality that hooks special keycodes. However, moving it towards SUBS() would likely make it simpler to add lots more by leveraging chorded activations without the need to add dummy keycodes.

tomsaleeba, 2021-02-20:
@jpconstantineau Yep, that's the layout for artsey.

I haven't seen the germ combo engine before but it sounds cool. Being able to run without dummy keycodes sounds like a win too. The QMK way of defining combos, when you use them like I do, doesn't feel the most concise as you have to define data in three different spots (keycode, the keys to press and handling code).

jpconstantineau, 2021-02-20:
They have a PR [here](https://github.com/qmk/qmk_firmware/pull/8591) to bring some of that stuff over.

tomsaleeba, 2021-02-20:
Oh, while we talking about existing engines, I found this [chording engine by dennytom](https://github.com/qmk/qmk_firmware/tree/master/users/dennytom/chording_engine). This engine seems to leverage the tap dance feature and relies on code generation to do the heavy lifting.

jpconstantineau, 2021-02-20:
@tomsaleeba I made progress on this...  I am pretty much at the same stage you are (but with simpler configuration). 
They can be triggered with 2,3,4 keys and can call any keycode and even macros.

I too am faced with the timing issues where I could get the trigger keys to show up on key-down or key-up...  artseyio is a good case to test the code as there are overlapping chords and without the timing figured out, it's really difficult to see if there are any overlap issues with the code; or if it's a timing issue...

jpconstantineau, 2021-03-27:
I have merged in the combo engine within develop and will be making a release to master soon.  I have it to a point where it's useful and can be used.  Behavior is a bit different that other firmwares as it looks for keycodes and not specific keypresses. 

There are 3 types of combos in place:

- Simple combo - replace 2 keycodes with another one
- Send string combo - send a string when the combo is activated (can replace macros)
- Send series of keycodes - Similar to sending a string but you also send special keys (arrows for example). This can replace more complex macros

I'll need feedback from people to see how it could be improved.  In the meantime, I need to move on to addressing other issues in other places...

closed 2021-03-27 by jpconstantineau