jpconstantineau/BlueMicro_BLE #155: Documentation: hardware_config pin assignment

spad, 2020-09-09:
I'm adding a simple 3x3 matrix keyboard for feather52840.
In the hardware_config.h

`#define MATRIX_ROW_PINS {25, 26, 27, 28}`
`#define MATRIX_COL_PINS {29, 30, 2, 3}`

the pin numbers what do they refer to on the hardware? 
On the NRF chip GPIOs are called PB0.x. 
Perhaps to the MQT52 module pins?
Or to the adafruit PINs (D1, D2 ...)

thanks

jpconstantineau, 2020-09-22:
For the feather52840, use the adafruit Pin numbers.
See the example [here](https://github.com/jpconstantineau/BlueMicro_BLE/blob/master/firmware/keyboards/breadboard-feather/hardware/feather52840/feathernrf52840/hardware_config.h).

More detailed information on the actual mapping used by the feather can be found in the [variant](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/variants/feather_nrf52840_express/variant.cpp#L26) in the BSP.
 
For all other boards (PCA10056), use this formula:
32*X + Y
where the GPIO is described as PX.Y
0.11 = 11
1.11 = 32+11 = 43

You can check the PCA10056 [definition](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/variants/pca10056/variant.cpp#L26) and it's a 1:1 pin to GPIO (unlike the feather52840)

jpconstantineau, 2020-10-25:
Hi Ben,  Can you comment on the changes to the documentation I did here: https://github.com/jpconstantineau/BlueMicro_BLE/blob/develop/docs/configuring_firmware.md

Any comments for improvements will be appreciated.

BenRoe, 2020-10-28:
@jpconstantineau thanks for making the detailed documentation.
I would add a calculation example at the end of this [section](https://github.com/jpconstantineau/BlueMicro_BLE/blob/develop/docs/configuring_firmware.md#all-other-modules-nrf52832-or-nrf52840). Then it should be 💯 clear.

> where the GPIO is described as PX.Y
0.11 = 11
1.11 = 32+11 = 43

jpconstantineau, 2021-01-28:
A fair bit of documentation has been done. See this [page](http://bluemicro.jpconstantineau.com/docs/understand_gpios) for details.

closed 2021-01-28 by jpconstantineau