jpconstantineau/BlueMicro_BLE #165: KC_MS_BTN1 doesn't make any sense

uyoyo117, 2020-10-08:
when using KC_MS_BTN1 mouse wont click, and some modifier get stuck preventing use of mouse (as if a transparent window is getting all mouse clicks ) 

**To Reproduce**
Steps to reproduce the behavior:
1. set any key as KC_MS_BTN1
2. use this key to click 
3. after it won't click multiple times try using a regular mouse 

happens on windows 10 
PS. where can i find DOC on mouse key codes ?

jpconstantineau, 2020-10-09:
Thanks for the comments. I am not satisfied with the current implementation of mouse functions. They are located in keyscanner.cpp (around line 600 where special keypresses are sorted out) and in bluetooth.cpp (function specific to mouse functions).     

I believe the main source of issues might be that either mouse HID or keyboard HID are sent but not both or multiple mouse entries at the same time. 

I would welcome any improvements to the implementation; even if it modifies base support from the adafruit library.

jpconstantineau, 2020-10-09:
Probably the only documentation I have for mouse keys is the code in bluetooth.cpp, lines 495 on...  The implementation is very simple.

uyoyo117, 2020-10-10:
so in order to use adafruit library correctly you need to use 
`blehid.mouseButtonPress(hid_conn_hdl, MOUSE_BUTTON_LEFT);`
as a toggle so when holding a button you call it once and then when the key is released you need to call 
`blehid.mouseButtonRelease(hid_conn_hdl);`
this is the way they intended to use it.
i simulated it using delay because i don't know the best way to implement it

jpconstantineau, 2020-10-12:
Thank you for the details. I will need to check on the current
implementation and update it accordingly. I suspect it’s a bit different
than you indicated

jpconstantineau, 2022-04-20:
he latest refactor now supports multiple mousekeys  to be pressed and they are all sent at the same time,  in the same report.

commit e91071ae (2022-04-20): refactored mouse keys. fixes #165 closes #294
closed 2022-04-20 by jpconstantineau