jpconstantineau/BlueMicro_BLE #169: Error on build (osx): <brace-enclosed initializer list>()

BenRoe, 2020-10-16:
**Describe the bug**
In the `BlueMicro_BLE/build/macos folder` i call this command `./build-macos -v crkbd` and get the follwoing error.

```bash
/Users/Ben/Git/BlueMicro_BLE/output/.build/sketch/keymap.cpp:43:1: error: could not convert '<brace-enclosed initializer list>()' from '<brace-enclosed initializer list>' to 'Key'
   43 | };
      | ^
      | |
      | <brace-enclosed initializer list>
```

I did not changed anything. 

**Desktop (please complete the following information):**
 - OS: OSX 10.14.6
 -Master Branch 

Here is the full output
```bash
BlueMicro_BLE/build/macos  master ✗                                                                                            
▶ ./build-macos -v crkbd 

-----------------------------------------------------------------------
   Arduino BlueMicro Build Script
-----------------------------------------------------------------------

Building crkbd:all:all adafruit:nrf52:feather52832:softdevice=s132v6

Checking file locations
-----------------------------------
Arduino Installation... OK
Arduino Data Location... OK
Adafruit nRF52 Package... OK

Compiling
-----------------------------------
crkbd:bcat:left adafruit:nrf52:feather52832:softdevice=s132v6... ===info ||| Progress {0} ||| [0.22]
===info ||| Progress {0} ||| [0.45]
===info ||| Progress {0} ||| [0.67]
===info ||| Progress {0} ||| [0.89]
===info ||| Progress {0} ||| [1.12]
===info ||| Progress {0} ||| [1.34]
===info ||| Progress {0} ||| [1.56]
===info ||| Progress {0} ||| [1.79]
===info ||| Progress {0} ||| [2.01]
===info ||| Progress {0} ||| [2.23]
===info ||| Progress {0} ||| [2.46]
===info ||| Progress {0} ||| [2.68]
===info ||| Progress {0} ||| [2.90]
===info ||| Progress {0} ||| [3.13]
===info ||| Progress {0} ||| [3.12]
===info ||| Progress {0} ||| [6.25]
===info ||| Progress {0} ||| [9.38]
===info ||| Progress {0} ||| [12.50]
===info ||| Progress {0} ||| [15.62]
===info ||| Progress {0} ||| [18.75]
===info ||| Progress {0} ||| [21.88]
===info ||| Progress {0} ||| [25.00]
===info ||| Progress {0} ||| [28.12]
===info ||| Progress {0} ||| [31.25]
===info ||| Progress {0} ||| [31.25]
===info ||| Progress {0} ||| [34.38]
===info ||| Progress {0} ||| [37.50]
===info ||| Progress {0} ||| [37.72]
===info ||| Progress {0} ||| [37.95]
===info ||| Progress {0} ||| [38.17]
===info ||| Progress {0} ||| [38.39]
===info ||| Progress {0} ||| [38.62]
===info ||| Progress {0} ||| [38.84]
===info ||| Progress {0} ||| [39.06]
===info ||| Progress {0} ||| [39.29]
===info ||| Progress {0} ||| [39.51]
===info ||| Progress {0} ||| [39.73]
===info ||| Progress {0} ||| [39.96]
===info ||| Progress {0} ||| [40.18]
===info ||| Progress {0} ||| [40.40]
===info ||| Progress {0} ||| [40.62]
/Users/Ben/Git/BlueMicro_BLE/output/.build/sketch/keymap.cpp:43:1: error: could not convert '<brace-enclosed initializer list>()' from '<brace-enclosed initializer list>' to 'Key'
   43 | };
      | ^
      | |
      | <brace-enclosed initializer list>
/Users/Ben/Git/BlueMicro_BLE/output/.build/sketch/keymap.cpp:43:1: error: could not convert '<brace-enclosed initializer list>()' from '<brace-enclosed initializer list>' to 'Key'
/Users/Ben/Git/BlueMicro_BLE/output/.build/sketch/keymap.cpp:43:1: error: could not convert '<brace-enclosed initializer list>()' from '<brace-enclosed initializer list>' to 'Key'
/Users/Ben/Git/BlueMicro_BLE/output/.build/sketch/keymap.cpp:43:1: error: could not convert '<brace-enclosed initializer list>()' from '<brace-enclosed initializer list>' to 'Key'
===info ||| Progress {0} ||| [25.00]
===info ||| Progress {0} ||| [50.00]
===info ||| Progress {0} ||| [75.00]
===info ||| Progress {0} ||| [100.00]
exit status 1
```

jpconstantineau, 2020-10-16:
Corne keyboard should not be able to use the feather or nrf52832. They are
physically larger than the space available for installing the pro micro.

This error is usually present when there is a problem with the keymap
files.

Did you change anything in the files you have?

jpconstantineau, 2020-10-17:
After some investigation, this is related to the macos script that's not yet been updated to match the new folder structure.
I will need help on this.  I can create a branch to get this started but for testing, I will need someone to confirm that it works fine.

BenRoe, 2020-10-17:
@jpconstantineau tested the updated script and `Successful: 250 Failed: 0`.

jpconstantineau, 2020-10-17:
Fixed with pull request 170.  Closing

closed 2020-10-17 by jpconstantineau