jpconstantineau/BlueMicro_BLE #177: Column issue with 2.1c

bacarp89, 2020-10-31:
I am working on a 5x12 keyboard utilizing the 2.1c, and for some reason a key press on the top row triggers like the whole column. I also replicated this on a second bluemicro off of the board I'm building, and it has the same effect with jumping the same pins. Do you have any insight on what may cause this?

Thanks!

jpconstantineau, 2020-10-31:
If it does the same thing regardless of the controller, I suspect it's a
keyboard build issue.  Have a look at the troubleshooting hints for the
ergotravel.

https://github.com/jpconstantineau/ErgoTravel/blob/master/Troubleshooting_steps.md

Hopefully this will help pinpoint the problem.

There is also a new serial cli that might help. Using the Arduino IDE,
connect to the board using the serial monitor and send h. You will see a
menu coming up. The different troubleshooting tools might help.

bacarp89, 2020-11-01:
Sorry for the delay, I went through those as well unfortunately, I was able to replicate the issue using the same pins on the contra firmware as well. when I use pin 12/B1, pin 3/F6 does not work properly.

jpconstantineau, 2020-11-02:
I don't have a BlueMicro 2.1c.  There is a BlueMicro2.1b and a BlueMicro2.0c.  Which one is it?  There is a minor difference between the two and it's about which GPIO D2 and D3 are connected to.  Both have B1 and F6 connected the same way. 

https://github.com/jpconstantineau/NRF52-Board/blob/master/BlueMicro2.1b/schematic.pdf
https://github.com/jpconstantineau/NRF52-Board/blob/master/BlueMicro2.0c/BluemicroV2.0c.pdf

Are you using the Adafruit Feather nrf52832 board as the selection in the IDE (or feather52832 as the fqbn in the build scripts)?

bacarp89, 2020-11-02:
Sorry about that, I am using a 2.1b currently, and yes, I have been using nrf52832 as the selection.

jpconstantineau, 2020-11-03:
Have you tried the new Serial Debug CLI?  Using the Arduino IDE and the serial monitor, connect to the keyboard and type "h". It will return a menu of options.  There are 3 different GPIO testers.  I need to document them but hopefully they will help you towards a solution.

These testers bypass any matrix scanning and go direct to the GPIOs and see what's connected to what.
**g  run GPIO Tester**: test each GPIO by reading the values when enabling pull-up and pull-down resistors. 
**m  full matrix gpio tester**: useful to help identify GPIO rows and columns when pressing keys.  Will accumulate the vectors as you press keys
**k  single key matrix gpio tester**: Similar to the full matrix tester but will only provide the GPIOs of pressed keys.

bacarp89, 2020-11-03:
I'm not sure how, but I just toasted this. I will pick this up after I
order a new one, thank you for all your help and patience, and sorry for my
lack of understanding around all of this!

closed 2020-11-03 by bacarp89