jpconstantineau/BlueMicro_BLE #178: Blue LED stays on when sleeping

torda385, 2020-11-05:
**Describe the bug**
Keyboard sleeps (assumed because it takes a few key presses to begin registering again) but the blue LED stays on.

**To Reproduce**
Flash with Arduino IDE 1.8.13. Wait for keyboard to sleep.

**Expected behavior**
Blue light should turn off.

**Desktop (please complete the following information):**
 - OS: Windows 10

**Additional context**
Have added #define BLE_LED_ACTIVE false to hardware_config.h but it made no difference

jpconstantineau, 2020-11-05:
Refactor of the Status LED logic should take care of this.  
This refactor is needed for the following reasons:
1 - Enhance support for boards with LEDs connected to other GPIOs
2 - Use other GPIOs to enable for external LEDs (as opposed  to on-board LEDs)
3 - Some boards have 0, 1 or 2 Status LEDs available.
4 - Adafruit will be deprecating support for Bluefruit.autoConnLed()

jpconstantineau, 2020-11-18:
PR #179  will take care of this.  It also adds more features to use the LEDs (Blue and red) properly.  All you need to define are the following (adjust for the GPIOs) 
	#define  STATUS_BLE_LED_PIN  19  //blue = 0.19
//	#define  BLE_LED_ACTIVE 1  // default = 1 when PIN is defined.
///	#define  BLE_LED_POLARITY 1 // default = 1 when PIN is defined.  

	#define  STATUS_KB_LED_PIN 17  //red = 0.17
//	#define  STATUS_KB_LED_ACTIVE 1  // default = 1 when PIN is defined.
//	#define  STATUS_KB_LED_POLARITY 1  // default = 1 when PIN is defined.

If you don't have a specific LED, don't define it. It won't try using it.
If you want to disable it, set ACTIVE to 0
If a 0 turns on the LED, put POLARITY to 0.  This is unusual but might happen...