jpconstantineau/BlueMicro_BLE #202: Link to *Edit this page* on the docs points to the wrong domain

tomsaleeba, 2021-01-28:
**Describe the bug**
The link to *Edit this page* on the docs points to the wrong domain.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to docs, e.g: http://bluemicro.jpconstantineau.com/docs/flash
2. Scroll to the bottom and check the URL for the *Edit this page* link

**Expected behavior**
It points to https://github.com/jpconstantineau/BlueMicro_BLE/edit/develop/documentation/docs/flashing_firmware.md or https://github.com/jpconstantineau/BlueMicro_BLE/edit/master/docs/flashing_firmware.md.

**Actual behaviour**
It points to https://github.com/facebook/docusaurus/edit/master/website/docs/flashing_firmware.md, that is, it's still using the Facebook Docusaurus repo.

jpconstantineau, 2021-01-28:
Indeed. good catch on that.  I'll have to find where that default and update it.

jpconstantineau, 2021-01-28:
Links are still broken but now point to a 404 on github.

tomsaleeba, 2021-01-28:
The link has the `docs` dir twice: https://github.com/jpconstantineau/BlueMicro_BLE/tree/documentation-docusaurus/documentation/docs/docs/flashing_firmware.md

If you remove on of them, it works.

jpconstantineau, 2021-01-29:
Verified the latest push and it's now linked properly.

closed 2021-01-29 by jpconstantineau

tomsaleeba, 2021-01-31:
It's fixed :heart_eyes: