jpconstantineau/BlueMicro_BLE #247: Volume Up/Down keys appear to be repeated

jamiehs, 2021-08-23:
This is the QMK key tester showing the issue (one tap up and one tap down):
[image]

The expected behavior is that the volume up and down keys should increment the volume once when tapped. In my case, I'm seeing between 4 and 10 presses being sent per 'tap' of the key. All other keys are working as expected.

I've verified this on Mac OS (Macbook Pro), and Windows 10. The keycodes being sent from the keymap are just:

`KC_AUDIO_VOL_UP` and `KC_AUDIO_VOL_DOWN`

I've built a keyboard definition for my custom Dactyl Manuform; wired as one single keyboard, and this is the full keymap shown. Everything else about the conversion to BlueMicro (it was formerly using a Feather 32u4 Bluefruit with QMK) is working well though! Just this one issue.

My `keymap.cpp` is just calling the codes directly. No macros, layers, or modifiers are involved.
```cpp
std::array<std::array<Key, MATRIX_COLS>, MATRIX_ROWS> matrix =
    {KEYMAP(
          KC_GRV, KC_1, KC_2, KC_3, KC_4, KC_5,                    KC_6, KC_7, KC_8, KC_9, KC_0, KC_MINS, KC_EQL,
          KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T,                    KC_Y, KC_U, KC_I, KC_O, KC_P, KC_LBRC, KC_RBRC,
         LAYER_1, KC_A, KC_S, KC_D, KC_F, KC_G,                    KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_BSLS,
       KC_LSHIFT, KC_Z, KC_X, KC_C, KC_V, KC_B,                    KC_N, KC_M, KC_COMM, KC_DOT, KC_SLSH, KC_NO, RALT(KC_F11),
          KC_LCTRL, KC_LGUI, KC_LALT, KC_ESC,                        KC_AUDIO_VOL_UP, KC_ENT, KC_SPC, KC_BSPC, 
                         KC_LSHIFT,                                  KC_AUDIO_VOL_DOWN, LAYER_1
)};
```

Happy to troubleshoot if there are any ideas I could try from this end.

jpconstantineau, 2021-08-23:
Volume UP/Down does seem to suffer from repeated keys.

Can you try  OS(KC_AUDIO_VOL_UP) and OS(KC_AUDIO_VOL_DOWN) instead of KC_AUDIO_VOL_UP and KC_AUDIO_VOL_DOWN and let me know how that works out?

jamiehs, 2021-08-24:
Hey, thanks for taking a look.
I just gave the one-shot function a try and there's no difference in the way they repeat.

If there's anything else you'd like me to try/tweak, just let me know.

jpconstantineau, 2021-08-24:
The only other thing I can think of right now is to move it to a "tap layer"...

Have a look at the example [here](https://github.com/jpconstantineau/BlueMicro_BLE/blob/master/firmware/keyboards/4x4Tutorials/keymaps/taphold/keymap.cpp) on how to do that.
'
In summary, you put KC_NO on the regular layer and put KC_AUDIO_VOL_UP on the tap layer.
Then you make sure you have the loop to add the tap layer to memory.

jamiehs, 2021-08-30:
As an additional data point, it seems that when the keyboard is wired in via USB (I'm now testing this on another board I'm migrating over that is using the nRF52840, the last one was a nrf52832) the repeated keys behave the same, but are sent much more rapidly, and thus there are more repeats.

Also, I tried the tap layer method. It _kind of works_.
With it enabled as below, the host will see either a single press, or two presses. The strange thing is that if one press is sent, the key behaves as if it's being held. If two presses are sent, each press has its own keyup and keydown events and it works fine. Additionally, if you repeatedly tap the key, it sends two presses consistently. If you tap the key only once, that seems to be the case where a sent key may behave as if it's stuck.

I hope some of this is helpful at least. 

```cpp
void setupKeymap() {

    uint32_t layer1[MATRIX_ROWS][MATRIX_COLS] =
    KEYMAP(
     _______, KC_F1,   KC_F2,   KC_F3,   KC_F4,     KC_F5,         KC_F6,   KC_F7,   KC_F8,   KC_F9,   KC_F10,  KC_F11,  KC_F12,
     _______, BLE_1,   BLE_2,   BLE_3,   _______, _______,         _______, _______, KC_UP,   _______, _______, _______, _______,
     _______, BATT,    _______, _______, _______, _______,          _______, KC_LEFT, KC_DOWN, KC_RIGHT,_______, _______, RESET,
     _______, BLEI,    _______, _______, _______, OUT_BT,          OUT_USB, _______, KC_LBRC, KC_RBRC, _______, _______, _______,
          _______, _______, _______, _______,                                _______, _______, _______, KC_DEL,
                         _______,                                                      _______, _______);

    uint32_t tap[MATRIX_ROWS][MATRIX_COLS] =
    KEYMAP(
     _______, _______, _______, _______, _______, _______,         _______, _______, _______, _______, _______, _______, _______,
     _______, _______, _______, _______, _______, _______,         _______, _______, _______, _______, _______, _______, _______,
     _______, _______, _______, _______, _______, _______,         _______, _______, _______, _______, _______, _______, _______,
     _______, _______, _______, _______, _______, _______,         _______, _______, _______, _______, _______, _______, _______,
          _______, _______, _______, _______,                              KC_AUDIO_VOL_UP , _______, _______, _______,
                         _______,                                                    KC_AUDIO_VOL_DOWN , _______);

    /*
     * add the other layers
     */
    for (int row = 0; row < MATRIX_ROWS; ++row)
    {
        for (int col = 0; col < MATRIX_COLS; ++col)
        {
            matrix[row][col].addActivation(_L1, Method::PRESS, layer1[row][col]);
            matrix[row][col].addActivation(_QWERTY, Method::MT_TAP, tap[row][col]);
        }
    }

}
```

jamiehs, 2021-08-30:
OK, another update.
I got it to work by changing the report type from repeating to non-repeating. But this has the side effect of not being able to hold the key down to send it repeatedly.

I can live with having to tap the volume keys for now, so I'll keep it this way. I do want to keep looking into this, so I'll let you know if I see another, less brute-force way to solve it.

(I just changed `MR` to `MK`)
```cpp
// firmware/hid_keycodes.h
#define KC_AUDIO_VOL_UP      MK(KC_T) //HID_USAGE_CONSUMER_VOLUME_INCREMENT 
#define KC_AUDIO_VOL_DOWN    MK(KC_U) //HID_USAGE_CONSUMER_VOLUME_DECREMENT  
```

closed 2021-08-30 by jamiehs