jpconstantineau/BlueMicro_BLE #250: [Feature Request] PMW3360 support for trackball builds

xnopasaranx, 2021-11-07:
I am in the process of building a dactyl manuform rtrack with a recess for a trackball and PMW3360 sensor. The person (https://gitlab.com/qurn) who made the design uses qmk and their keyboard is wired. I would like to use the bluemicro firmware and was wondering if there was a way to also implement a trackball hid device or at least map mouse movements somehow.

As I have all hardware for the build coming in right now, I could help trying stuff out if someone was to guide me a little.

#### additional info
this is the sensor breakout board in question and the wiring diagram for qmk and pro micros
[https://gitlab.com/keyboards1/dm_r_track/-/blob/master/dm_PMW_wire_diagram.png](url)

they have also begun to implement the sensor in a qmk fork here:
[https://gitlab.com/keyboards1/qmk_firmware/-/tree/master/keyboards/handwired/dactyl_manuform_r_track](url)

jpconstantineau, 2021-11-07:
BlueMicro_BLE uses Arduino at its core.  I would be interested in bringing support for pointing devices to the firmware (nunchuck, potentionemers, trackpads, etc) but need help in doing so (it's mostly a matter of time).  As such, I would help guide you in implementing this within BlueMicro_BLE.  My USB/BLE refactor will also help in making this possible as the mouse, keyboard and consumer HID messages will be better exposed.

First step would be a simple implementation within a simple arduino program.
This will serve as the proof of concept and test code to make sure that the core PWM3360 code needed would work.

From the proof of concept, the code can then be integrated within the BlueMicro_BLE firmware.

The proof of concept is important because porting AVR code to the nRF52 platform may appear simple but there won't be a "#include <avr/pgmspace.h>" available since this is avr specific.

There is already a PWM3360 arduino library out there ready to be used.  The examples provided in the library would probably have to be ported to the nRF52.  I suspect it would only be a matter of "merging" the examples from the library with the ones from either TinyUSB or BlueFruit (part of the nRF52 BSP).

xnopasaranx, 2021-11-09:
Great news, thanks for the quick response. My nice!nanos will arrive today
or tomorrow and I will try to contribute and work on a proof of concept.
The author of the qmk dactyl firmware with pwm3360 support has mentioned
working on an implementation with bluetooth support (I have to assume this
is actually bluemicro_ble) so I will try to contact them aswell, maybe they
already have this working.

Mit freundlichen Grüßen,
Tilman Oestereich @Mausbrand

mausbrand
Informationssysteme GmbH

Tel. +49(0) 231 - 108 7222 31 | ***@***.***
Speicherstrasse 33 | 44147 Dortmund Hafen
Geschäftsführung Roland Brose & Felix Frank
HRB 26959, Amtsgericht Dortmund | USt-IdNr. DE282929360

xnopasaranx, 2021-11-09:
yikes, I never answered on a github issue via email. I took out the company adress and my real name, silly me ^^.
I have also contacted qurn on reddit and asked them for any code they have already.

xnopasaranx, 2021-11-27:
qurn came back to me via reddit and informed me that they gave up on it. but I picked up my pmw3360 from the dhl shop today. I am almost finished handwiring the left side of my dactyl manuform r-track and will start on the right side soon, so I'll try to play with the sensor and firmware before I put them inside and see if I can get them to talk with each other

xia0, 2021-11-30:
Please keep us updated. Interested in seeing how this goes!

jpconstantineau, 2022-01-27:
Any updates on how you got the sensor going on non-AVR code?  If I remember well the AVR code had lots of PROGMEM that aren't really portable to ARM chips.

xnopasaranx, 2022-01-27:
No update as of yet. I have switched to ZMK to test my rgb per key lighting and have the keyboard working perfectly over BLE now. ZMK also has a branch that supports mousekeys and someone has implemented the pimoroni trackball as a proof of concept on their own fork. Apparently someone else got their pmw3360 working with that code, switching out the pimoroni  zephyr drivers (they have a pmw3360 driver in their SDK), but the licenses are incompatible and the code is private because of that. So that is where I am at right now, trying to figure out how this could be adapted for use here and how to get around those licensing issues...

efenderbosch, 2022-06-04:
Any updates @xnopasaranx ? I'm going to start a dactyl manuform w/ trackball build soon and I'd love to have BT. Right now it looks like none of the 3 big players, QMK, BlueMicro or ZMK, can do both of those at the same time without using a private fork that's likely far behind main.

EricDracula, 2022-06-08:
Yes....Currently, I think zmk may be the most possible one to realize the wireless trackball.

jpconstantineau, 2022-06-08:
I have been working very slowly on the refactoring of BlueMicro to enable integrating pointing devices a whole lot easier using a common HID library I have created. 

If the PMW3389 code can work with the Arduino Framework with this [library](https://github.com/jpconstantineau/BlueMicro_HID_Arduino_Library), one would be able to use multiple processors with it. (RP2040, nRF52xxx, SAMD21, SAMD51) with no or very little code change.  If anyone wants to give it a try, there is sample code that comes with the library that only needs the code for interacting with the hardware to be added.  Once that is done, I can help in merging it in the BlueMicro Firmware - and potentially making it into a dedicated library for others to use.

efenderbosch, 2022-06-09:
I *think* that the PMW3389 should be fully compatible w/ PMW3360, but I'm not positive. Mine's en route, along w/ my nice! nanos.

Here's Arduino examples for each:
- https://github.com/mrjohnk/PMW3360DM-T2QU/tree/master/Arduino%20Examples
- https://github.com/mrjohnk/PMW3389DM/tree/master/Arduino%20Examples