jpconstantineau/BlueMicro_BLE #274: debug the project

billionfish, 2022-03-04:
hi I am very interested in your work. can you give any suggestion on debug?
For nrf52 I would like to know if there is a way to debug the code in a single step? Can this project be converted to a 'SEGGER Embedded Studio for ARM' project for debugging? Is there some tutorial to guide me to convert the project for single step debugging in ide like vscode? thanks!

jpconstantineau, 2022-03-07:
As the project is built on the Adafruit nfr52 Arduino Board Support Package, I suspect there isn't any way to migrate it to Segger Studio.  In fact, when I first created the project, I started looking at seggeer studio and thee nrf52 sdk and after hours of work, I still wasn't able to get something successfully built.  With the Adafruit nrf52 Arduino framework, I was able to get going within a few minutes.

The usual way things are debugged in the Arduino IDE can be applied to visual studio.  We don't use standard debugging tools (the softdevice create hard faults when we hit breakpoints - due to rigid timing requirements from the BLE) but only use serial.print statements.  Thee Arduino IDE integrate a serial monitor from within the IDE.  With VS code, you need to install an extension for connecting to serial ports.

closed 2022-03-07 by jpconstantineau

billionfish, 2022-03-09:
thank you for your reply