jpconstantineau/BlueMicro_BLE #325: Direct Pin connection keyboard support

rootexpression, 2024-01-18:
Is there a way to do direct pin?  I didn't find anything other than matrix configuration in the docs, or in my searches (I may have done a bad job of choosing the right keywords, so if I missed something please let me know).

For really compact keyboards and macro pads it'd be much easier to, in some cases, just direct wire them, and help keep it low profile in some instances (by a few mm).

I'd like to see that as an option, if its something that can find its way onto the todo list.

jpconstantineau, 2024-01-18:
Yes, there is. I don't recall an example but they use a dummy pin as a row
and use the individual pins for  where the switches are connected. (or vice
versa).

rootexpression, 2024-01-18:
By dummy pin, do you mean set a pin that doesn't exist on the board?  Like P0.99? Or a pin that exists, but isn't connected?

jpconstantineau, 2024-01-18:
Not quite. 0 I believe is usually connected to a crystal and will ignore
any I/O to it. Or a pic that the chip doesn't have connected to an outside
pin on the board. This will depend on the board in question

rootexpression, 2024-01-18:
Excellent info to have.  Thank you!  I'll have to dig through the schematics of the boards I have and see what I can turn up.  I'm betting they all have at least a few pins connected to nothing.  I can give that a try when I get something wired up to test with.